ros-navigation/navigation2 #3819: Record random seeds used by AMCL

dimaxano, 2023-09-14:
## Feature request

#### Feature description
It is not a secret that AMCL is not quite a deterministic algorithm. But sometimes it is required to take robot data and reproduce locally (i.e. run amcl locally based on data recorded in rosbag) the situation happened during operation. Indeterminism becomes essential here. Sometimes I'm just unable to reproduce certain mislocalization cases locally after multiple retries.

One of the sources of such indeterminism (along with multithreading) is random value generation functions. 

What we can do about that (aka feature description): save random seed values (publish them to `/rosout`) used by AMCL in the current run. So next time we need to reproduce the given AMCL run, we can feed the seed value from that run in the past.

#### Implementation considerations
I've brought that feature to our (Coalescent Mobile Robotics) fork of nav2, so we are good with bringing that feature to upstream (with or without modification) if you find it useful:
- cmrobotics/navigation2#46

SteveMacenski, 2023-09-14:
AMCL is in maintenance only mode for bugs and will not receive further developments as a legacy system. We're talking internally right now about a replacement but that's still ~6 months out from being ready for beta testing (which if you're interested in, let me know!)

So closing as `wontfix` since it will receive no development time. However, if you were to implement it, we would be willing to field a PR and discuss there.

closed 2023-09-14 by SteveMacenski

dimaxano, 2023-09-18:
1. Yep, we are willing to be among the beta testers of the new localization functionality!
2. I'll make a PR and you can check if it is suitable