ros-navigation/navigation2 #3828: Add support for macOS

srmainwaring, 2023-09-21:
## Feature request

#### Feature description

Support macOS (intel and arm64), libraries compile on macOS using clang and run natively.

#### Implementation considerations

There a number of fairly minor changes required to support macOS 13.4, Xcode 14.3.1.

This branch contains general changes for clang broken out by sub-library to make it easier to cherry-pick changes into a PR.

- https://github.com/srmainwaring/navigation2/tree/prs/pr-humble-clang

This branch contains a couple of additional changes specific to macOS on arm64, they're not suitable for merging as compilation options have just been disabled if they don't work on macOS.

- https://github.com/srmainwaring/navigation2/tree/wips/wip-humble-macos

Both branches target `humble` as this is the version of ROS 2 I have a native build for on macOS. Can re-target `main` for a PR if this feature request is accepted.

Figure: rviz running nav2 / humble on macOS, testing off-board control using XRCE-DDS in ArduPilot. 
[image]

SteveMacenski, 2023-09-21:
> There a number of fairly minor changes required to support macOS 13.4, Xcode 14.3.1.

Please open a PR! Main is the right place, but we can immediately backport to Humble/Iron if we don't break ABI/API.

srmainwaring, 2023-09-21:
Would you prefer one PR with all the changes, or a number of smaller ones broken out by component?

SteveMacenski, 2023-09-21:
All at once is fine as long as its not like 500+ lines. If its _alot_ of changes, break into modules sounds good

srmainwaring, 2023-09-22:
Submitted the PR. I'm still working on a port of ROS 2 Rolling to macOS, so I've not verified all the changes are good when compiling the `main` branch against that version. Humble is verified to be good on both Intel and M1 macs. I'll update the PR comment as I complete the rolling port.

SteveMacenski, 2023-09-28:
Handling it in the PR, looks good to me so closing this ticket since its broadly done except a couple small items

closed 2023-09-28 by SteveMacenski