ros-navigation/navigation2 #3833: [map_io]: Failed processing YAML file at position (-1:-1) for reason: bad file:

deepak-anzo, 2023-09-23:
## Bug report

**Required Info:**
[map_server-1]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1695461208.106685418] [map_server]: Creating
[lifecycle_manager-3] [INFO] [1695461210.377527778] [lifecycle_manager_localization]: Starting managed nodes bringup...                                                                               
[lifecycle_manager-3] [INFO] [1695461210.377714329] [lifecycle_manager_localization]: Configuring map_server                                                                                          
[map_server-1] [INFO] [1695461210.378863884] [map_server]: Configuring
[map_server-1] [INFO] [map_io]: Loading yaml file: /home/anzo/AMR/ros2/install/amr_navigation/share/amr_navigation/map/map_sep6_v3c.yaml
[map_server-1] [ERROR] [map_io]: Failed processing YAML file /home/anzo/AMR/ros2/install/amr_navigation/share/amr_navigation/map/map_sep6_v3c.yaml at position (-1:-1) for reason: bad file: /home/anzo/AMR/ros2/install/amr_navigation/share/amr_navigation/map/map_sep6_v3c.yaml
[map_server-1] [ERROR] [1695461210.379373559] []: Caught exception in callback for transition 10
[map_server-1] [ERROR] [1695461210.379404708] []: Original error: Failed to load map yaml file: /home/anzo/AMR/ros2/install/amr_navigation/share/amr_navigation/map/map_sep6_v3c.yaml
[map_server-1] [WARN] [1695461210.379458940] []: Error occurred while doing error handling.
[map_server-1] [FATAL] [1695461210.379483807] [map_server]: Lifecycle node map_server does not have error state implemented
[lifecycle_manager-3] [ERROR] [1695461210.819169855] [lifecycle_manager_localization]: Failed to change state for node: map_server
[lifecycle_manager-3] [ERROR] [1695461210.819251944] [lifecycle_manager_localization]: Failed to bring up all requested nodes. Aborting bringup.

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-09-23:
What is the file? Is it properly formatted? Please show it here

SteveMacenski, 2023-09-28:
@deepak-anzo please provide more information so we can resolve your problem. We'll close this ticket in 1 week if we don't hear back. Feel free to comment if there's still an issue

closed 2023-10-04 by SteveMacenski

stevedanomodolor, 2024-06-20:
Getting this now when running the updown test. It seems to manifest there only for me though, 

`-> [map_server-5] [ERROR] [map_io]: Failed processing YAML file /opt/overlay_ws/install/nav2_bringup/share/nav2_bringup/launch/maps/tb3_sandbox.yaml at position (-1:-1) for reason: bad file: /opt/overlay_ws/install/nav2_bringup/share/nav2_bringup/launch/maps/tb3_sandbox.yaml`

MansiP441, 2024-07-12:
Check if the provided path and yaml_filename are correct.