ros-navigation/navigation2 #3838: robot_navigator.py

c185scmyw, 2023-09-26:
## Bug report
In nav2_simple_commander package, there is a file named "robot_navigator.py". In that file line 429, "self" should be deleted. 

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-09-26:
https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/robot_navigator.py#L429

There is no self in the `main` branch's version. What version are you referring to? Please send an actual link or blob with more detail about what you think is wrong

SteveMacenski, 2023-09-28:
I also checked for Iron / Humble, the only other two active distros and don't see an error on that line. Closing. Happy to resolve some issue if you can point it out more specifically, but I don't see a problem nor a file with L429 having self on it

closed 2023-09-28 by SteveMacenski