ros-navigation/navigation2 #3840: Can not mirror the nav2_bringup package

GPrathap, 2023-09-27:
I want to make some modifications to the [nav2_bringup][1] package. Therefore, I just copied the nav2_bringup package put it in the workspace and ran the [Turtlebot 3][2] simulator. The following error occurred. Without copies, no errors, all work fine. 

https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/README.md

   `[ERROR] [launch]: Caught exception in launch (see debug for traceback): cannot import 
   name 'PushROSNamespace' from 'launch_ros.actions' (/opt/ros/humble/lib/python3.10/site- 
   packages/launch_ros/actions/__init__.py)`

  [1]: https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/README.md
  [2]: https://navigation.ros.org/getting_started/index.html#getting-started

## Bug report
**Required Info:**

- Operating System:
  - OS and version  Ubuntu 22.04.
- ROS2 Version:
  - Humble

SteveMacenski, 2023-09-27:
Did you copy from the Humble or main branch? main targets rolling. I believe this is your issue 😉

closed 2023-09-27 by SteveMacenski

icosac, 2023-10-25:
Hi, I'm having a similar trouble. I'm following [this](https://navigation.ros.org/getting_started/index.html), so I installed the versions of the packages listed in the guide for Humble, which is the ROS2 version I have, but when I try to launch the file I obtain the same error on the substitution. 

Indeed, if I look at the documentation for `nav2_common.launch.ReplaceString`, there is no `condition` field.

GPrathap, 2023-10-26:
Ensure that all branches are humble,

c00lum, 2025-03-06:
I fixed mine with:

`sudo apt upgrade`