ros-navigation/navigation2 #3848: Problem not to run halt correctly in bt_action_node

CihatAltiparmak, 2023-10-01:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
- ROS2 Version: Rolling
- Version or commit hash: 42670c0d7bbe3712afde83550a3747fac1f19c4a
- Package: nav2_behavior_tree

#### Description of the debug

When `bt_action_node` is halted before goal is completed, it is wait that the goal cancels but not wait that corresponding `result_callback` finish its job. This brings about not to run `return_callback` on time. 

#### Expected behavior
When `bt_action_node` is halted, if corresponding goal is not completed, it should be cancelled goal and triggered `on_cancelled` callback as soon as `result_callback` of corresponding goal.

#### Actual behavior
When `bt_action_node` is halted, if corresponding goal is not completed, it's cancelled goal but not triggered `on_cancelled` callback as soon as `result_callback` of corresponding goal.

#### Additional information
For more information about this bug, please look at [this discussion](https://github.com/BehaviorTree/BehaviorTree.ROS2/issues/27)

I think it should be added new test case for the scenairo to halt `bt_action_node` before goal execution is completed.

I tested on `ros::rolling` docker image.

#### My question on this bug

https://github.com/ros-planning/navigation2/blob/42670c0d7bbe3712afde83550a3747fac1f19c4a/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L363-366

Is this part still necessary after this bug is solved? I think about it but i haven't found it.

SteveMacenski, 2023-10-02:
Great to point out and I appreciate the test case! I've responded in your PR

CihatAltiparmak, 2023-10-02:
Hello @SteveMacenski , i have commited according to your feedback. Thanks for feedback.

closed 2023-10-02 by SteveMacenski