ros-navigation/navigation2 #3856: Local Planner struggling to follow the global plan

Rak-r, 2023-10-03:
## Question/Help wanted

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
     -ROS2 Humble

- Version or commit hash:
  -binary

#### Steps to reproduce issue
I am using gazebo garden with ROS2 humble and setting up the NAV2 stack. Using launch files `navigation.launch.py` and `localization.launch.py`  from the official nav2 humble branch. To test on a simple env, I created a simple world and mapped it using slam toolbox.

#### Actual behavior

When I set the NAV2GOAL in rviz, the global plan displayed and robot initially starts following it. However just after reaching the goal and when an angular goal is set, sometimes the robot tries to follow the path but mostly becomes clueless and freezes at the position. However,I can see the spin behaviour of front wheels on the same position and backup behaviour.
 
# My Terminal logs
`[controller_server-1] [INFO] [1696337764.532271886] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337765.531261962] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337766.531246666] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337767.531247922] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337768.531247092] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337769.531248888] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1696337775.731550355] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1696337775.731595022] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1696337775.760733930] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-1] [INFO] [1696337775.761549825] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [INFO] [1696337776.761690222] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1696337777.761686957] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1696337785.961727920] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1696337785.962205388] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[behavior_server-3] [INFO] [1696337785.981072063] [behavior_server]: Running spin
[behavior_server-3] [INFO] [1696337785.981606049] [behavior_server]: Turning 1.57 for spin behavior`

# Video demonstration
[Screencast from 03-10-23 14:24:20.webm](https://github.com/ros-planning/navigation2/assets/85680564/9f194047-0bd0-4517-8e2d-38a633d434af)

# Controller server and Plannar server params

`
controller_server:
  ros__parameters:
   
    use_sim_time: False
    controller_frequency: 5.0
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.5
    min_theta_velocity_threshold: 0.001
    progress_checker_plugin: "progress_checker"
    goal_checker_plugins: ["general_goal_checker"]
    controller_plugins: ["FollowPath"]
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    general_goal_checker:
      stateful: True
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.001
      yaw_goal_tolerance: 0.001
    # DWB parameters
    FollowPath:
      plugin: "dwb_core::DWBLocalPlanner"
      debug_trajectory_details: True
      min_vel_x: 0.0
      min_vel_y: 0.0
      max_vel_x: 0.26
      max_vel_y: 0.26
      max_vel_theta: 1.0
      min_speed_xy: 0.0
      max_speed_xy: 0.26
      min_speed_theta: 0.0
      acc_lim_x: 2.5
      acc_lim_y: 2.5
      acc_lim_theta: 3.2
      decel_lim_x: -2.5
      decel_lim_y: -2.5
      decel_lim_theta: -3.2
      vx_samples: 20
      vy_samples: 5
      vtheta_samples: 20
      sim_time: 1.7
      linear_granularity: 0.05
      angular_granularity: 0.025
      transform_tolerance: 0.2
      xy_goal_tolerance: 0.25
      trans_stopped_velocity: 0.1
      short_circuit_trajectory_evaluation: True
      stateful: True
      critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
      BaseObstacle.scale: 0.02
      PathAlign.scale: 32.0
      PathAlign.forward_point_distance: 0.1
      GoalAlign.scale: 24.0
      GoalAlign.forward_point_distance: 0.1
      PathDist.scale: 32.0
      GoalDist.scale: 24.0
      RotateToGoal.scale: 32.0
      RotateToGoal.slowing_factor: 5.0
      RotateToGoal.lookahead_time: -1.0
      publish_trajectories: true
      publish_local_plan: true
controller_server_rclcpp_node:
  ros__parameters:
    use_sim_time: False
'

'planner_server:
  ros__parameters:
    planner_plugins: ["GridBased"]
    use_sim_time: False

    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerHybrid"
      downsample_costmap: false           # whether or not to downsample the map
      downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      tolerance: 0.25                     # dist-to-goal heuristic cost (distance) for valid tolerance endpoints if exact goal cannot be found.
      allow_unknown: true                 # allow traveling in unknown space
      max_iterations: 1000000             # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000    # Maximum number of iterations after within tolerances to continue to try to find exact solution
      max_planning_time: 5.0              # max time in s for planner to plan, smooth
      motion_model_for_search: "REEDS_SHEPP"    # Hybrid-A* Dubin, Redds-Shepp
      angle_quantization_bins: 72         # Number of angle bins for search
      analytic_expansion_ratio: 3.5       # The ratio to attempt analytic expansions during search for final approach.
      analytic_expansion_max_length: 3.0  # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting
      minimum_turning_radius: 0.5        # minimum turning radius in m of path / vehicle
      reverse_penalty: 2.0                # Penalty to apply if motion is reversing, must be => 1
      change_penalty: 0.0                 # Penalty to apply if motion is changing directions (L to R), must be >= 0
      non_straight_penalty: 1.2           # Penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 2.0                   # Penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      retrospective_penalty: 0.015
      lookup_table_size: 20.0             # Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false     # Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.
      smooth_path: True                   # If true, does a simple and quick smoothing post-processing to the path
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1.0e-10
        do_refinement: true
        refinement_num: 2    
planner_server_rclcpp_node:
  ros__parameters:
    use_sim_time: False`

#### Additional information
I made custom nodes to convert gazebo information of scan and odometry to ROS2 time as I wanted everything to work on single time source which makes sense for the real robot implementation as well.
I tried making some changes with costmaps params and some other transform tolerances and frequency related params but still similar results.

SteveMacenski, 2023-10-03:
I generally don't provide configuration help, but this one is relatively obvious

>   xy_goal_tolerance: 0.001
>   yaw_goal_tolerance: 0.001

Your goal tolerances are absurdly low. 1mm. I highly doubt your localization is 1mm in quality for that to be remotely reasonable :laughing: Set that to be more reasonable and you should be good to go (beyond refinement tuning)

closed 2023-10-03 by SteveMacenski

Rak-r, 2023-10-03:
@SteveMacenski  I tried to set both goal tolerances to 0.1  but still  the same situation. I don't have much idea about these params and I tried to search around a lot and looked at  the configuration guide but still the issue persists.  I know its very trivial issue I might facing. Could you advice?

SteveMacenski, 2023-10-03:
The issue tracker is not the right place for advice, try Robotics Stack Exchange (new ROS Answers) or the Navigation configuration guides (navigation.ros.org)

Rak-r, 2023-10-03:
@SteveMacenski  I have opened an query on robotics stack exchange.

[(https://robotics.stackexchange.com/questions/104488/local-planner-struggling-to-follow-the-global-plan)]