ros-navigation/navigation2 #3857: Inputs and parameters in BT nodes

nlamprian, 2023-10-04:
I tried using the navigate through poses action today. Our robots have links that are prefixed by the robot name. I was getting errors about failed transforms with target frame map or base_link. I kept trying to figure out what parameters I was not setting. I had to find out the hard way that the errors were coming from getCurrentPose, which was called by RemovePassedGoals, where the frames are passed as inputs.

Issue 1: the global and robot base frames, which are normally defined as parameters across nav2, are inputs here. The consequence, in this case, is that a BT file is now only applicable to one particular instance of a robot.

The same and other BT nodes also declare ROS parameters. At runtime, these parameters exist under the autogenerated BT navigator nodes, bt_navigator_node_navigate_{to_pose,through_poses}_rclcpp_node.

Issue 2: If a user wants to set a parameter of a BT node, they need to do so in a node that they have not explicitly run themselves, which is unintuitive.

Have these issues been brought up before? What are your thoughts or suggestions?
When do you consider a 'variable' to be an input and when to be a parameter?
Should there be an option to pass parameters from the BT navigator down to the BT nodes? (for instance, the aforementioned frames are already defined as parameters in the BT navigator)

nlamprian, 2023-10-04:
Sorry about that :( I made the arbitrary assumption that these topics wouldn't have received any changes, so I skipped the requested info.

Issue 1: Excellent!

Issue 2: I'm referring to [this](https://github.com/ros-planning/navigation2/blob/humble/nav2_bringup/params/nav2_params.yaml#L103). The first time I needed to set a BT node parameter, I instinctively sought to put it under bt_navigator. I had to find out in practice that this is not the case.

Sidenote: I also noticed that the navigators are now parameterized, which is my other big wish :partying_face:

From the looks of it, it doesn't seem these changes would be coming to humble. But anyway, it's amazing to see all this active development.

SteveMacenski, 2023-10-04:
Issue 2 is also solved in Rolling - albeit only in recent weeks 
- https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_server_impl.hpp#L138

I literally copy all the params from the BT navigator over to that client node that is accessible in the BT so that you can set them under the BT Navigator namespace and they'll be gettable anywhere in that node / BT / etc :smile: 

Yeah, no chance for humble since these are majorly breaking changes. I tried actually backporting to Iron and couldn't there either :/ So only in Rolling or once Jazzy comes out. Though, perhaps Iron could be done. Would that be useful to you?

> But anyway, it's amazing to see all this active development.

Welcome to the pool, the water's fine :beach_umbrella: We'd love your code contributions, community involvement, or company's financial support to help keep it going!

closed 2023-10-04 by SteveMacenski

nlamprian, 2023-10-04:
> Perhaps Iron could be done. Would that be useful to you?

We have no plans to move to Iron. Thanks for offering.

SteveMacenski, 2023-10-05:
I love these kinds of tickets: already solved :laughing: