ros-navigation/navigation2 #3873: Planner benchmark not functional (buggy)

Manouselis, 2023-10-12:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04 (jammy) 
- ROS2 Version:
  -  ros2 humble
- Version or commit hash:
  - Commit hash: 9e63d4349c2d53302bb41435beba7e323400fbf1

#### Steps to reproduce issue

Follow README instructions in [navigation2/tools/planner_benchmarking](https://github.com/ros-planning/navigation2/tree/main/tools/planner_benchmarking)

#### Expected behavior
>Run a set of planners over randomly generated maps, with randomly generated goals for objective benchmarking.

#### Actual behavior
NAV2 stack is not initialized (properly). More specifically, I get the following warnings and errors:
```
[map_server-1] [ERROR] [map_io]: Failed processing YAML file /home/ros2_oa/install/nav2_bringup/share/nav2_bringup/maps/map.yaml at position (-1:-1) for reason: bad file: /home/ros2_oa/install/nav2_bringup/share/nav2_bringup/maps/map.yaml
[map_server-1] [ERROR] [1697121114.123512494] []: Caught exception in callback for transition 10
[map_server-1] [ERROR] [1697121114.123517150] []: Original error: Failed to load map yaml file: /home/ros2_oa/install/nav2_bringup/share/nav2_bringup/maps/map.yaml
[map_server-1] [WARN] [1697121114.123524238] []: Error occurred while doing error handling.
[map_server-1] [FATAL] [1697121114.123529140] [map_server]: Lifecycle node map_server does not have error state implemented
[lifecycle_manager-5] [ERROR] [1697121114.123858557] [lifecycle_manager]: Failed to change state for node: map_server
[lifecycle_manager-5] [ERROR] [1697121114.123879620] [lifecycle_manager]: Failed to bring up all requested nodes. Aborting bringup.
```

I suspect that the problem arises in the following line of code in the `planning_benchmark_bringup.py` file:
` map_file = os.path.join(nav2_bringup_dir, 'maps', 'map.yaml')` (line 26)
I believe this happens because the `map.yaml` file does not exist in that directory anymore.

#### Additional information
If I disregard the errors and warnings and proceed to follow the rest of the instructions (i.e., run `metrics.py`), I get the following: 
`[INFO] [1697121448.525634049] [basic_navigator]: change map service not available, waiting...`

SteveMacenski, 2023-10-12:
There is no such thing as `map.yaml` https://github.com/ros-planning/navigation2/tree/main/nav2_bringup/maps

It also addresses that same yaml file in the bringup for the benchmark https://github.com/ros-planning/navigation2/blob/main/tools/planner_benchmarking/planning_benchmark_bringup.py#L26

How did you obtain `map.yaml`?  I don't see that in the codebase

SteveMacenski, 2023-10-13:
Ah I see it in the Humble branch… that’s interesting. I’ll need to look into that! Please confirm though that’s where you see it so I know I’m barking up the right tree

Manouselis, 2023-10-13:
Yes, my ROS2 version is humble

SteveMacenski, 2023-10-13:
https://github.com/ros-planning/navigation2/commit/5b87fabadbdb5ad6c1cc56066e77e4de1a8b26df The default is updated if you pull in that update locally.

Did you follow the instructions in the readme so that you have all the planners? Did you configure them for the behavior you'd like them to have? This is not a zero-setup solution, you need to specify what tests you'd like to configure the planners and which planners you'd like to test. 

https://github.com/ros-planning/navigation2/tree/humble/tools/planner_benchmarking#planning-benchmark

Manouselis, 2023-10-16:
I can confirm that after configuring the planners, the `planner_benchmarking` now works.

SteveMacenski, 2023-10-16:
I imagine its the same analog problems. Configure them.

closed 2023-10-16 by SteveMacenski

Manouselis, 2023-10-17:
> I imagine its the same analog problems. Configure them.

Excuse my naivete, but even configured it does not seem to work with the following parameters:
```
planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    use_sim_time: True
    planner_plugins: ["SmacHybrid"]
    SmacHybrid:
      plugin: "nav2_smac_planner/SmacPlannerHybrid"
      tolerance: 0.5
      motion_model_for_search: "DUBIN" # default, non-reverse motion
      smooth_path: false # should be disabled for experiment
      analytic_expansion_max_length: 0.3 # decreased to avoid robot jerking

smoother_server:
  ros__parameters:
    smoother_plugins: ["simple_smoother", "constrained_smoother"]
    simple_smoother:
      plugin: "nav2_smoother::SimpleSmoother"
    constrained_smoother:
      plugin: "nav2_constrained_smoother/ConstrainedSmoother"
      w_smooth: 1.0 # tuned
```
Have you tried the code (for Humble) and it works for you Steven?