ros-navigation/navigation2 #3883: heap-use-after-free in nav2_costmap_2d_core

Rrooach, 2023-10-15:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
  - 
- ROS2 Version: Humble
  - 
- Version or commit hash: 4b9bd8638de2cdd5bde363c84ecd4760105ec00e
  - 
- DDS implementation: FASTRTPS
  - 

#### Steps to reproduce issue
Basically, I installed ros-humble on a remote server, compiled it with AddressSanitizer, And when I start to boot the Navigator2 system and try to fuzz testing it, I get the following ASAN report (https://pastebin.com/DLyEGk9x).

#### Expected behavior
The program run without any problem
#### Actual behavior
Crashed with ASAN report
#### Additional information

So, I compile the project using 
```shell
CC=clang CXX=clang++  colcon build --parallel-workers 32  --cmake-args -DCMAKE_C_FLAGS=" -g  -w  -fsanitize=address " -DCMAKE_CXX_FLAGS="  -g  -w -fsanitize=address "  -DBUILD_TESTING=OFF
```
I start the fuzzing process, and I get the error report: https://pastebin.com/DLyEGk9x

It seems that there is an incorrect def-use chain between test_planner_is_path_valid (freed here )and libnav2_costmap_2d_core (allocated here).
It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

SteveMacenski, 2023-10-16:
I'm a little preoccupied with ROSCon this week, mind opening a PR to resolve any issues that these bring up? That would be an appreciate contribution! Luckily the thing causing the issue is a unit test, not a production system, so that's good news at least that this isn't a 5-alarm fire issue :sweat_smile: 

Steve

SteveMacenski, 2023-10-28:
https://github.com/ros-planning/navigation2/pull/3910 Resolved!

closed 2023-10-28 by SteveMacenski