ros-navigation/navigation2 #3886: Costmap resolution parameter not working as expected

LeifHookedWireless, 2023-10-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu
- ROS2 Version:
  - Iron with Nav2 source
- Version or commit hash:
  - Nav2 461a7ba1d296
- DDS implementation:
  - Cyclone DDS

#### Steps to reproduce issue

```
Set resolution for local and global costmap in your nav2_params.yaml
 resolution: 0.025
Load a map with lower resolution. e.g. 0.0125
```

#### Expected behavior
Static layers of both local and global costmaps should be resized and resampled to the param file resolution.
I would expect this to be done when the map is loaded, e.g. `StaticLayer::processMap()`

#### Actual behavior
Parameter has effect on global costmap. 
Its resolution is overwritten by the incoming map.

local costmap resolution is not overwritten, it keeps the parameter value.
local costmap is resampled in `StaticLayer::updateCosts()`, but the resampling is sparse so in this case only 25% of the map is sampled, and much of the static layer is missed resulting in large holes in the local costmap.
See image.
[image]

#### Additional information

For historical reasons, we use a fine 1.25 cm costmap, but for navigation, we would like the performance gain of a larger costmap, e.g. 2.5 cm.
We expected the `resolution` parameter to do that.

#### Implementation considerations
I have a local fix that resamples correctly in `StaticLayer::updateCosts()` which fixes the costmap for us for now (see image), but I don't think this the right fix.
[image]

I looks like the map is supposed to be resampled in `StaticLayer::processMap()` but that functionality is just missing.

LeifHookedWireless, 2023-10-16:
I know 1.25 cm is too small, that's why I changed the `resolution` parameter. 
Sounds like you're saying that won't help (would be nice if it did...)
I have to change the map instead?

SteveMacenski, 2023-10-17:
> I do think dynamic sample would be a cool feature, so I will work on it.

In that case, you can leave the ticket open! This also seems like the type of ticket someone in the future might come along and find