ros-navigation/navigation2 #3890: Wrong name in NavigateThroughPosesNavigator comment

Alexbeast-CN, 2023-10-19:
in [https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/navigators/navigate_through_poses.hpp](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/include/nav2_bt_navigator/navigators/navigate_through_poses.hpp)

```cpp
/**
 * @class NavigateToPoseNavigator
 * @brief A navigator for navigating to a a bunch of intermediary poses
 */
class NavigateThroughPosesNavigator
```

`@class NavigateToPoseNavigator` should be `@class NavigateThroughPosesNavigator`

SteveMacenski, 2023-10-23:
Fixed - thanks! In general, you can just open PRs for things like this, you don't need to file a ticket!

closed 2023-10-23 by SteveMacenski

Alexbeast-CN, 2023-10-24:
I'll try to open a PR next time.

Cheers!