ros-navigation/navigation2 #3898: "1 package failed: nav2_mppi_controller" while building Nav2 package

Rudresh172, 2023-10-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble (both binary packages and built from source)
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```
source /opt/ros/humble/setup.bash
sudo apt install \
  ros-$ROS_DISTRO-navigation2 \
  ros-$ROS_DISTRO-nav2-bringup \
  ros-$ROS_DISTRO-turtlebot3*

source /opt/ros/humble/setup.bash
mkdir -p ~/nav2_ws/src && cd ~/nav2_ws
git clone https://github.com/ros-planning/navigation2.git --branch $ROS_DISTRO ./src/navigation2
rosdep install -y \
  --from-paths ./src \
  --ignore-src
colcon build \
  --symlink-install
```

#### Expected behavior
All packages installed successfully

#### Actual behavior
Cloning into './src/navigation2'...
remote: Enumerating objects: 39313, done.
remote: Counting objects: 100% (1773/1773), done.
remote: Compressing objects: 100% (845/845), done.
remote: Total 39313 (delta 1136), reused 1265 (delta 862), pack-reused 37540
Receiving objects: 100% (39313/39313), 88.88 MiB | 5.46 MiB/s, done.
Resolving deltas: 100% (26133/26133), done.
executing command [sudo -H apt-get install -y ros-humble-test-msgs]
[sudo] password for rudresh: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  ros-humble-test-msgs
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 468 kB of archives.
After this operation, 7,807 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-test-msgs amd64 1.2.1-1jammy.20230919.204708 [468 kB]
Fetched 468 kB in 6s (78.5 kB/s)              
Selecting previously unselected package ros-humble-test-msgs.
(Reading database ... 319302 files and directories currently installed.)
Preparing to unpack .../ros-humble-test-msgs_1.2.1-1jammy.20230919.204708_amd64.deb ...
Unpacking ros-humble-test-msgs (1.2.1-1jammy.20230919.204708) ...
Setting up ros-humble-test-msgs (1.2.1-1jammy.20230919.204708) ...
Processing triggers for libc-bin (2.35-0ubuntu3.4) ...
executing command [sudo -H apt-get install -y python3-zmq]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  python3-zmq
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 283 kB of archives.
After this operation, 1,464 kB of additional disk space will be used.
[... 194 lines cut ...]
   57 |   auto node = std::make_shared<rclcpp_lifecycle::LifecycleNode>("node");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
/home/rudresh/nav2_ws/src/navigation2/nav2_mppi_controller/test/noise_generator_test.cpp:59:3: error: ‘ParametersHandler’ was not declared in this scope
   59 |   ParametersHandler handler(node);
      |   ^~~~~~~~~~~~~~~~~
/home/rudresh/nav2_ws/src/navigation2/nav2_mppi_controller/test/noise_generator_test.cpp:81:55: error: ‘handler’ was not declared in this scope
   81 |   generator.initialize(settings, false, "test_name", &handler);
      |                                                       ^~~~~~~
gmake[2]: *** [test/CMakeFiles/noise_generator_test.dir/build.make:76: test/CMakeFiles/noise_generator_test.dir/noise_generator_test.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:334: test/CMakeFiles/noise_generator_test.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_mppi_controller [3min 47s, exited with code 2]
Aborted  <<< nav2_rotation_shim_controller [53.6s]

Summary: 33 packages finished [8min 8s]
  1 package failed: nav2_mppi_controller
  1 package aborted: nav2_rotation_shim_controller
  2 packages had stderr output: nav2_mppi_controller nav2_simple_commander
  3 packages not processed

#### Additional information
I followed instructions from - https://navigation.ros.org/development_guides/build_docs/index.html
https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html#install-development-tools-and-ros-tools

SteveMacenski, 2023-10-23:
Clean rebuild by deleting your install / build spaces and trying again clean. If you have binaries installed, remove them when building MPPI from source until the next sync comes about.

Rudresh172, 2023-10-24:
Thanks for the quick response, removing binaries and rebuilding worked!
Thanks!

closed 2023-10-24 by Rudresh172

ymei, 2023-12-30:
I encountered the same problem on identical Ubuntu and ROS2 versions.  My observation is that packages installed via `apt`: `ros-$ROS_DISTRO-nav2-*`, need to be removed from the system in order for local compilation of the git repo (the humble branch) to succeed.