ros-navigation/navigation2 #3908: Should we enable topic statistics in Navigation 2 by modifying source code? [Question]

alanxuefei, 2023-10-28:
ROS 2 provides a feature for collecting topic statistics as detailed in the [Topic Statistics Tutorial](https://docs.ros.org/en/foxy/Tutorials/Advanced/Topic-Statistics-Tutorial/Topic-Statistics-Tutorial.html). This feature could provide valuable insights into the performance and behavior of navigation tasks, enhancing debugging and monitoring capabilities.

The statistics collected include message period (interval between consecutive messages) and message age (time from message creation to receipt). For each of these metrics, the mean, maximum, minimum, variance, and count values are computed.

As an example,  [odom_subscriber](https://github.com/ros-planning/navigation2/blob/c0fd78fce326fc40c127dca26058fa079d34940d/nav2_dwb_controller/nav_2d_utils/include/nav_2d_utils/odom_subscriber.hpp#L73C5-L77C78) does not enable this feature, which requires setting the `.topic_stats_options` in `create_subscription`.

Below is the existing code snippet:

```cpp
odom_sub_ =
  nh->create_subscription<nav_msgs::msg::Odometry>(
  odom_topic,
  rclcpp::SystemDefaultsQoS(),
  std::bind(&OdomSubscriber::odomCallback, this, std::placeholders::_1));
```

The feature could be activated within the Navigation2 code base, however, not all Nav2 users might find this necessary. Hence, users who require this feature have the option to modify and compile the source code, as opposed to installing the binary version. Does it make sense?

SteveMacenski, 2023-10-28:
I would be perfectly happy with enabling this where-ever it would be reasonably useful. Could you provide an exhaustive set of topics that you'd be interested in analyzing? Obviously some things like the action calls might not be overly interesting considering they could be called irregularly and are a bit of inside-mechanics. 

From a quick glance, I note:
- Odometry subscriptions
- Subscriptions to the Scan, PC2, Range in the Collision Monitor
- costmap / footprint subscribers?
- Velocity smoother cmd_vel
- Various BT nodes
- Range, voxel, obstacle, etc processing costmap layers 

Once we have a list and they're enabled, its worth doing some high level performance testing just to make sure it doesn't make some massive impact on runtime performance / CPU utilization (which I wouldn't expect) and we can enable it by default.

Sound like a plan? Hopefully that's something you could contribute!

alanxuefei, 2023-10-29:
The listed considerations are significant for thorough monitoring:
1. **Behavior Tree (BT) Nodes Variance**.
2. **Transform (TF) Tree Updates**.
3. **Subscriber Message Receipts** (ROS 2 exclusively offers integrated statistics measurement for subscribers).

Focusing initially on the third point, the following topics that are subscribed to are worth monitoring:

If we focus on point 3 first, the following subscribed topics are worth to be monitored. 
| Node Name                   | Subscribers                                |
|-----------------------------|--------------------------------------------|
| `planner_server`            | `/clock`                                   |
| `global_costmap/global_costmap` | `/clock`, `/scan`                      |
| `controller_server`         | `/clock`, `/odom`                          |
| `local_costmap/local_costmap` | `/clock`, `/scan`                       |
| `smoother_server`           | `/clock`, `/global_costmap/costmap_raw`    |
| `velocity_smoother`         | `/clock`, `/cmd_vel_nav`                   |
| `collision_monitor`         | `/scan`                                    |

In ROS1, users activate topic statistics through parameter settings during runtime. You can see this feature in [here](https://github.com/ros/ros_comm/blob/030e132884d613e49a576d4339f0b8ec6f75d2d8/clients/roscpp/src/libros/statistics.cpp#L60).

However, ROS2 operates differently — it only allows users to enable topic statistics during startup, specifically when using the 'create_subscription' function along with the 'option' field.

Considering this, a potential design to facilitate topic statistics within navigation2 could be to allow users to set parameters as shown below:

To enable topic statistics in navigation2, one pretiental design is that the user may just set parameters as below.
```
controller_server:
  ros__parameters:
    topic_statistics:
      odom:
        publish_period_sec: 1  # Set the publishing period (in seconds)
        statistics_topic: "/odom_statistics"  # Set the topic name for the publication of the statistics
```
The changes of codes may be as below. The 'getSubscriptionOptions' function is added.

```
class OdomSubscriber
{
public:
  explicit OdomSubscriber(nav2_util::LifecycleNode::SharedPtr nh, std::string default_topic = "odom")
  {
    nav2_util::declare_parameter_if_not_declared(
      nh, "odom_topic", rclcpp::ParameterValue(default_topic));

    std::string odom_topic;
    nh->get_parameter_or("odom_topic", odom_topic, default_topic);

    // Get SubscriptionOptions from a dedicated function
    auto options = getSubscriptionOptions(nh, odom_topic);

    // Create the subscription with the configured options
    odom_sub_ = nh->create_subscription<nav_msgs::msg::Odometry>(
      odom_topic,
      rclcpp::SystemDefaultsQoS(),
      std::bind(&OdomSubscriber::odomCallback, this, std::placeholders::_1),
      options);
  }

  rclcpp::SubscriptionOptions getSubscriptionOptions(nav2_util::LifecycleNode::SharedPtr nh, const std::string& odom_topic)
  {
    // Extract the topic name (e.g., "odom" from "/odom" or "ns/odom")
    auto topic_name_start = odom_topic.find_last_of('/') + 1;
    std::string topic_name = odom_topic.substr(topic_name_start);

    // Construct the parameter names based on the topic name
    std::string publish_period_param = "topic_statistics/" + topic_name + "/publish_period_sec";
    std::string statistics_topic_param = "topic_statistics/" + topic_name + "/statistics_topic";

    // Create a SubscriptionOptions object
    auto options = rclcpp::SubscriptionOptions();

    // Check if the parameters are declared
    bool parameters_declared = nh->has_parameter(publish_period_param) &&
                               nh->has_parameter(statistics_topic_param);

    if (parameters_declared)
    {
      int publish_period_sec;
      nh->get_parameter(publish_period_param, publish_period_sec);
      std::string statistics_topic;
      nh->get_parameter(statistics_topic_param, statistics_topic);

      // Enable topic statistics
      options.topic_stats_options.state = rclcpp::TopicStatisticsState::Enable;

      // Set the publishing period (converted from seconds to std::chrono::seconds)
      options.topic_stats_options.publish_period = std::chrono::seconds(publish_period_sec);

      // Set the statistics publication topic name
      options.topic_stats_options.publish_topic = statistics_topic;
    }

    return options;
  }
```

What are the steps to create a branch for contributing? Should I utilize git commands locally to create a branch and then push it to the nav2 repository?

SteveMacenski, 2023-10-30:
Create a branch on your fork and submit a draft PR when you want a review -- but like I mentioned above, its important to first before anything else identify all topics that are valuable to monitor, I would not merge in any PR that is not a complete implementation that covers the important bases. I started a thread on top. I don't think `/clock` is, since we don't have any subscribers to `/clock` within Nav2 directly (indirectly, sure, but that's not something we can control here).

alanxuefei, 2023-11-01:
The  `/clock ` has been removed. We may proceed with  `/scan ` and  `/odom ` after narrowing the list of topics. 
The reasons for monitoring these topics are as below:

- High CPU usage could freeze the nodes that are publishing  `/scan ` and  `/odom `, causing Nav2 to malfunction.
- High latency in '/scan' could impact the performance of collision_monitor.  

An open question remains: Is it worth adding extra code to measure topics directly at the exact reception point inside Nav2, as opposed to monitoring them externally via the command line using ros2 topic hz and ros2 topic delay?  

If the answer is "Yes", I would like to contribute to this effort.

SteveMacenski, 2023-11-02:
> Is it worth adding extra code to measure topics directly at the exact reception point inside Nav2

I think the answer is a strong "maybe", which I know isn't a satisfying answer. If you're saying you need _this_ instead of externally, then perhaps others do as well. The biggest question I want to know is the overhead -- if it negligible, then I figure why not. If its major, then it needs a real reason. I don't expect it'll be major, but I'd reserve any statement about "is it worth it..." until we have something to benchmark to know. 

I would though want it to be enabled by any particular topic exhaustively that it would be useful for, which is why I wanted to get a list going. It would also be nice if we merge it in to have some tool(s) that use that info so its not just enabled, but can provide some useful insight. What are you planning on doing with it and would that make sense as a Nav2 Util, a `/tools/` tool or something to live in the Lifecycle Manager as an watchdog?

alanxuefei, 2023-11-11:
A tool in `tools` could be similar to the commands of `ros2 topic hz` and `ros2 topic delay`. GPT is capable of easily generating a script.

We might consider adding a feature to the lifecycle manager. If topics (`/odom` and `/scan`) are not published at the expected frequency or experience delay, we could pause the robot until the system returns to normal. Through the lifecycle manager, we would transition all nodes into inactive states and then restore them to active states. Does this sound like a plan?

SteveMacenski, 2023-11-11:
I think that's hitting feature creep for the lifecycle manager to have that feature that is tied to specific topics - since subscribers need to be typed and perhaps users have other topic name(s) or type(s) that they want to watchdog. I think that should be a separate node that uses the lifecycle manager services to cycle down / up as necessary. 

That does sound like a useful feature. I would be very open to that contribution to take in `N` topics or some statistics topic - or both to be really optional and might be useful for testing and if it breaks out of a particular band of normal operations it brings down the lifecycle until its restored. That should be relatively straight forward.

alanxuefei, 2023-11-14:
nav2_watchdog could monitor (subscribe to) the topics flowing within Navigation2.

One starting point could be monitoring for missed deadlines on the controller. The discussion at https://discourse.ros.org/t/nav2-soft-realtime-controller-server-option-available/34384 highlights the importance of the controller's real-time performance. nav2_watchdog could begin by monitoring the frequency (Hz) of /cmd_vel published by the controller and later expand to other topics, such as odom, subscribed to by the controller.

SteveMacenski, 2023-11-14:
Do the topic statistics publish on a global topic (e.g. `/statistics`)? If so, then I'd think it'd make more sense to use that instead to simply this watchdog using the best-practices tools. Subscribing directly to the topics have the added complexity of handling future user-defined types, so it would need to be recompiled from source. Instead, if the watchdog accumulation happens on-node, we have much simpler logic and only one callback to manage.

It would still be wise to have that node have a configured list of topics/nodes that its watching, in case one crashes so the self reports stop coming in at all. And so that you can select which you want to care about and which you do not.

SteveMacenski, 2024-01-10:
@alanxuefei any updates or progress here?

alanxuefei, 2024-03-16:
Close the issue first. We will reopen it when we have a new idea or if this feature becomes necessary.

closed 2024-03-16 by alanxuefei

TanayK07, 2026-10-16:
ran ros2_pulse on a composed nav2 bringup like i said i would. jazzy, loopback sim, use_composition:=True, 4 NavigateToPose goals (all succeeded). nothing changed in nav2, just LD_PRELOAD on the launch.

for the one component_container_isolated process:
- 18 nodes and 115 pubs/subs, each counted separately and tagged with its node, so a stalled node shows up even though it shares the process with 17 others
- /tf taken at 100 Hz by 8 separate subscriptions in that process
- /scan 10 Hz into both costmaps and collision_monitor, worst gap ~107 ms
- cmd_vel chain controller -> velocity_smoother -> collision_monitor at 20 Hz while driving

it all comes out as one jsonl line per window, which is roughly what i'd want to feed rosgraph_monitor as an always on health check.

run + per node table: https://github.com/TanayK07/ros2_pulse/actions/runs/38897785668