ros-navigation/navigation2 #3920: Make BT Action, Cancel, and Service nodes have configurable wait times

SteveMacenski, 2023-10-30:
Currently default at `1s` hardcoded, which should be sufficient for any well engineered system. However, users have asked for this to be configurable and we should oblige

SteveMacenski, 2023-10-31:
- https://github.com/ros-planning/navigation2/blob/363b55af3a61ccdd2f429689a174e49fb7cf7523/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L96
- https://github.com/ros-planning/navigation2/blob/363b55af3a61ccdd2f429689a174e49fb7cf7523/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L80
- https://github.com/ros-planning/navigation2/blob/363b55af3a61ccdd2f429689a174e49fb7cf7523/nav2_behavior_tree/include/nav2_behavior_tree/bt_cancel_action_node.hpp#L92

embeddedadam, 2023-11-08:
@SteveMacenski I would like to take care of this ticket.

SteveMacenski, 2023-11-08:
Great!! Let me know if you have any questions!

SteveMacenski, 2023-11-20:
Done! Thanks @embeddedadam !

closed 2023-11-20 by SteveMacenski