ros-navigation/navigation2 #3923: Update docs to better communicate dynamic footprint support

SteveMacenski, 2023-10-30:
Footprints can be updated dynamically depending on arm pose, load being carried, etc. Document this explicitly and visibly so that folks are aware (and what topics to publish to).

 CC @tonynajjar have any input here to share given you do this as I'm looking to better document it for the community?

tonynajjar, 2023-10-30:
Do I understand correctly that this ticket is to document custom solution built to enable dynamic footprints? If yes, why? How is this useful to the community?

(Or are you gathering requirements for the implementation of a Nav2 generic way)

SteveMacenski, 2023-10-30:
The goal is to document that its even possible at all to do this and how to do it (e.g. what topics to publish to, perhaps some high level guidance about how to compute the updated footprints). It came up in the User Survey that several people asked for this feature that already exists that they were not aware of. So I want to document it more explicitly and clearly that this is possible / easy to do

tonynajjar, 2023-10-30:
@jplapp @HovorunB friendly ping as you've worked on this more closely than I have, feel free to put some thoughts in.

david-wb, 2024-04-08:
Hello, I am trying to do exactly this, and a tutorial would have been helpful. Is it as simple as publishing a polygon to a single topic? If you could let me know the ROS message type for publishing a polygon and the topic name to use, I happy to help write a brief tutorial/demo on dynamically updating a robot's footprint.