ros-navigation/navigation2 #3925: nav2_bringup does notwork - ROS2 Humble

hjorthchristian, 2023-11-01:
- Operating System:
  Ubuntu 22.04
- ROS2 Version:
  Humble

I've downloaded the navigation2 and nav2- bringup  and turtlebot 3 packages with

sudo apt install ros-humble-navigation2
sudo apt install ros-humble-nav2-bringup
sudo apt install ros-humble-turtlebot3-gazebo

When I try to run:
ros2 launch nav2_bringup bringup_launch.py

It gives me the following error:
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: ReplaceString.__init__() got an unexpected keyword argument 'condition'

I've seen a similar problem in the closed discussions, however I have downloaded everything for the Humble distribution, and it still does not work. Any ideas why?
Thanks in advance! :)

SteveMacenski, 2023-11-01:
Try 

```
ros2 launch nav2_bringup tb3_simulation_launch.py
```

On for size. `bringup_launch.py` is not the entry point - its an internal launch file. Checkout the getting started guide https://navigation.ros.org/getting_started/index.html. I just tested the binaries and works for me!

closed 2023-11-01 by SteveMacenski

hjorthchristian, 2023-11-01:
> Try
> 
> ```
> ros2 launch nav2_bringup tb3_simulation_launch.py
> ```
> 
> On for size. `bringup_launch.py` is not the entry point - its an internal launch file. Checkout the getting started guide https://navigation.ros.org/getting_started/index.html. I just tested the binaries and works for me!

I've tried to run your command and it still gives the same error message, I've removed the packages and then I've tried again after redownloading the packages and it still gives me the same error neither of the commands work and both give the same output error. 
ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: ReplaceString.init() got an unexpected keyword argument 'condition'
@SteveMacenski  have you downloaded the latest versions of the packages or do you have another one? Thanks for your very quick response btw, it is much appreciated! :)

SteveMacenski, 2023-11-01:
I did an apt update/upgrade before testing this to make sure -- it seems to work on my side fine with Ubuntu 22.04 / Humble binaries. 

Do you also have a source build locally you're using / sourcing? Perhaps there's some cross pollination?

SteveMacenski, 2023-11-01:
I don't either -- it works fine on my end. Perhaps try `apt update/upgrade`?

hjorthchristian, 2023-11-01:
> I don't either -- it works fine on my end. Perhaps try `apt update/upgrade`?

@SteveMacenski  Thank you very much , it worked now! :) Is there any particular reason to it even though I've just donwloaded the latest packages?

SteveMacenski, 2023-11-01:
I have no idea and its probably not worth the time for me to figure it out haha