ros-navigation/navigation2 #3931: navfn_planner GoalOccupied exception

ladianchad, 2023-11-02:
## Bug report

**Required Info:**

- Operating System:
  - 22.04
- ROS2 Version:
  - humble
- Version or commit hash:
  - source da7f85d
- DDS implementation:
  - rmw_cyclone

#### Steps to reproduce issue
send goal like below image
[image]

#### Expected behavior
as my understand of navfn planner within context
https://github.com/ros-planning/navigation2/blob/da7f85d79dc06945a8c416070c37f7bfa39ea4ee/nav2_navfn_planner/src/navfn_planner.cpp#L297-L311
 plan success. with lethal goal. see below image.
[image]

#### Actual behavior
navfn palnner throw GoalOccupied
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2023-11-02:
I don't understand this ticket. Please provide detail about the error, what you expect to happen, what is happening, and what behavior you expect. This is not verbose enough for me to understand what the issue is.

SteveMacenski, 2023-11-16:
Closing due to lack of response and PR doesn't seem to solve a problem that I can identify. Feel free to comment here or on the PR and we can continue.

closed 2023-11-16 by SteveMacenski