ros-navigation/navigation2 #3956: Costmap2DPublisher refactoring

emSko, 2023-11-10:
* breaking the occupancy grid update population into its own function similar to `createCostmapUpdateMsg` called `createGridUpdateMsg`
* Remove code redundancy with msg creation
* Checking if keeping a `grid_` and a `costmap_raw_` as members is a necessity

_Originally posted by @SteveMacenski in https://github.com/ros-planning/navigation2/pull/3948#discussion_r1388298321_

SteveMacenski, 2023-11-10:
@emSko I think the first point you can do in your PR. The second isa nice to haves if you want to open follow up PRs, but the third point I don't think is practical, we probably do need both for the publisher since they serve different purposes.

emSko, 2023-11-15:
@SteveMacenski  The third bullet point refers to the class members declared here: https://github.com/ros-planning/navigation2/blob/c5eddb3d31914338549493ea315732c52342f4a3/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_publisher.hpp#L173 and here https://github.com/ros-planning/navigation2/blob/c5eddb3d31914338549493ea315732c52342f4a3/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d_publisher.hpp#L174 

I don't want to remove the functionality of sending both types of messages (costmap and grid) but rather I would like to unify the way the messages are created (as proposed in #3948  using `createCostmapUpdateMsg()` method ) by adding respective create methods and removing the unnecessary class members of `Costmap2DPublisher`.

I wanted to create a new issue for that because the previous PR started to become quite big.

It does not have very high priority, but since I have been working on this class recently I thought that I could do it.

SteveMacenski, 2023-11-15:
I don't think its so long - if you wanted to add `createGridUpdateMsg` you could :-)

closed 2023-11-21 by SteveMacenski