ros-navigation/navigation2 #3959: Uniformize polygon parameter type between costmap footprint and collision monitor

doisyg, 2023-11-13:
## Feature request

#### Feature description
Uniformize polygon parameter type between costmap footprint and collision monitor.
- Costmap example: https://github.com/ros-planning/navigation2/blob/e066ed1dee18daad2c6a95a49e5db4fc6407f77e/nav2_costmap_2d/launch/example_params.yaml#L35
- Collision monitor example: https://github.com/ros-planning/navigation2/blob/e066ed1dee18daad2c6a95a49e5db4fc6407f77e/nav2_collision_monitor/params/collision_monitor_params.yaml#L20

SteveMacenski, 2023-11-13:
Yes, good point

SteveMacenski, 2023-11-29:
I agree, please do! :heart: 

You should move the array parser to the nav2_utils package - that'll handle that concern :-)

closed 2023-12-23 by SteveMacenski