ros-navigation/navigation2 #3968: A good way to improve the overall obstacle avoidance ability of DWA

dahuacai, 2023-11-17:
[image]

When the global path is blocked by a large range of obstacles or blocked directly in front of a protruding obstacle, if DWA uses the original path to calculate the forward point, the generated trajectory will sometimes bypass the obstacle and then be pulled back near the obstacle.

This is because using the original path to find the forward point, the forward point may be occupied by obstacles, dwa can not track the forward point well. In other words, if there are many obstacles near the original path, the DWA cannot globally avoid these complex obstacles.

I find a good method to improve the global obstacle avoidance ability of DWA. When we detect that the original path transform_path is occupied by obstacles by using the local cost map, we traverse from the original path end to the starting point in the local cost map and find a free point. And use astar to compute a new path as transform_path

Using this new path to guide DWA to carry out forward-looking point calculation, then DWA can find an optimized obstacle avoidance trajectory with only a little amount of computation. At the same time, because the local cost map window is generally small, the cost of using local cost map for astar planning will be small. I have used the above ideas to transform DWA, and the measured results are very good.

As shown in the figure above, the original path is a blue line. When the original path is blocked by obstacles in the local cost map window, I will use astar to find a new path with brown lines, and then use DWA to calculate a new looking point, black points and dwa tracks, and green thick lines to avoid various obstacles.

SteveMacenski, 2023-11-17:
This seems like it might solve a problem but also create others. DWB is just a critic scoring algorithm at its core, so the way that you tune the critics or what critics you include have a pretty outsized impact on behavior. 

For instance, you could tune DWB with the path alignment relatively weak, so that we generally follow the path, but weight the obstacle critic more substantially. In that case, there would be no need for the mechanism that you've described because the robot has the obstacle critic that would make it perform similar maneuvers without the explicit intermediary planning stage. 

This seems to be trying most significantly to be 'getting around' fine tuning for your particular application. Or, perhaps a misunderstanding about how DWAs work in being objective scorers. There's no rule / law / standard that says DWBs are trying to uniquely follow a global path only. A reasonably well tuned DWB I would expect to be able to handle that situation without a problem for the example laid out above. 

With that said, to be fair, honestly doesn't seem like a bad idea. DWA/DWB are a real pain in the butt to tune. The way that this is implemented though is quite important. Its not good enough to simply use A*, it should be using the same underlying planning techniques that the planners themselves are utilizing. In your example, it looks like a differential drive robot without any serious constraints. Now imagine it was a car-like vehicle with ackermann constraints or using a planner to create optimal behavior. A naive A* would create possibly impossible behaviors that break the constraints of the vehicle. So in this sub-planning problem, we should be calling the planning server with a specific algorithm in mind to make sure we remain in compliance. 

The details here are quite important to potentially contribute back to upstream.

dahuacai, 2023-11-17:
code use astar to help dwb to avoid obstacle:
https://github.com/ros-planning/navigation2/issues/3967#issuecomment-1815701452

dahuacai, 2023-11-17:
In fact, I only use two evaluation indicators, BaseObstacleCritic obstacle avoidance and GoalDistCritic end alignment, and the obstacle avoidance weight is very low, as long as a free passable path can be generated. And path end alignment ensures that I go as far as possible to the desired global path point rather than going around somewhere else

SteveMacenski, 2023-11-17:
The weight being low in DWB might not mean that its not influential. DWB (annoyingly) doesn't have their metrics in a consistent unit or normalized scale. But, if the impact is that way, I might recommend tuning the inverse: obstacle critic has more power to override path tracking. That may be why you see the behavior you do. If there's not high costs in the way, then the obstacle critic wouldn't try to divert so you'd still be scoring based on the path tracking

SteveMacenski, 2023-11-21:
From discussion in #3967, closing due to A* solution not being a contribution for DWB codebase

closed 2023-11-21 by SteveMacenski