ros-navigation/navigation2 #3970: navigation_launch.py does not use params_file parameter?

slowrunner, 2023-11-20:
## Bug report?

- Operating System: Ubuntu 22.04.3  
- ROS2 Version: Humble apt install   
- Version or commit hash:  ros-humble-navigation2	1.1.12-1jammy.20231006.063129  
- DDS implementation: default  
  - dpkg -l | grep fastrt   
```
ii  ros-humble-fastrtps                               2.6.6-1jammy.20230919.195413            arm64        *eprosima Fast DDS* (formerly Fast RTPS) is a C++ implementation of the DDS (Data Distribution Service) standard of the OMG (Object Management Group).
```

#### Steps to reproduce issue  

I want to lower some velocities by passing a local my_nav2_params.yaml file, which is a modified copy of /opt/ros/humble/share/nav2_bringup/params/nav2_params.yaml.  

```
$ diff my_nav2_params.yaml /opt/ros/humble/share/nav2_bringup/params/nav2_params.yaml 
145,147c145
<       # changed for GoPiGo3
<       # max_vel_x: 0.26
<       max_vel_x: 0.1
---
>       max_vel_x: 0.26
151,153c149
<       # changed for GoPiGo3
<       # max_speed_xy: 0.26
<       max_speed_xy: 0.1
---
>       max_speed_xy: 0.26
346,351c342,343
<     # changed for GoPiGo3
<     # max_velocity: [0.26, 0.0, 1.0]
<     max_velocity: [0.1, 0.0, 1.0]
<     # changed for GoPiGo3
<     # min_velocity: [-0.26, 0.0, -1.0]
<     min_velocity: [-0.1, 0.0, -1.0]
---
>     max_velocity: [0.26, 0.0, 1.0]
>     min_velocity: [-0.26, 0.0, -1.0]
```

This does not do the trick:  

```
ros2 launch nav2_bringup navigation_launch.py params_file:=/home/ubuntu/HumbleDave/ros2ws/my_nav2_params.yaml
```

[The docs suggest this is the correct command](https://navigation.ros.org/setup_guides/footprint/setup_footprint.html#build-run-and-verification)  

**Is there a different launch file for what I want to do, or a different launch command string that will use my local modified copy of the nav2_params.yaml file?**  

#### Expected behavior  
- Navigating to a goal will proceed at a maximum of 0.1m/s  

#### Actual behavior  
- Navigation to a goal proceeds at 0.26 m/s (stopping too fast will tip robot over)  
 
#### Additional information  

I do not want to edit the default nav2_params.yaml, nor put my_nav2_params.yaml in /opt/ros/humble/share...  I want to use the my_nav2_params.yaml file in my ~/ros2ws/

SteveMacenski, 2023-11-20:
I see you previously filed on robotics stack exchange. People do not crosspost.

closed 2023-11-20 by SteveMacenski

slowrunner, 2023-11-20:
I posted on robotics stack exchange when I thought it was my inexperience with the package.  Then I found in the exact command I was using in the nav2 package documentation, which led me to believe this is a defect, not my lack of knowledge, thus "People do not cross post", unless further evidence suggests this is the correct place for the question.