ros-navigation/navigation2 #3982: NAV2 General Tutorial Problem on ROS 2 Humble

Cristian-wp, 2023-11-22:
**Required Info:**

- Operating System:
  - Ubuntu 22
- ROS2 Version:
  - Humble
- Version or commit hash:
  - git clone --recursive https://github.com/ros-planning/navigation2.git -b humble

#### Steps to reproduce issue
Hello I am Trying to follow [Nav2 General Tutorial](https://navigation.ros.org/tutorials/index.html) but I am stuck on the 2° and 3° one (same error).
I have following [(SLAM) Navigating While Mapping ](https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html) tutorial, so I have modify the  .../nav2_ws/src/navigation2/nav2_bringup/params/nav2_params.yaml file as you can see below
```
global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.22
      resolution: 0.05
      track_unknown_space: true
      plugins: ["static_layer", "obstacle_layer", "inflation_layer","stvl_layer"]
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: scan
        scan:
          topic: /scan
          max_obstacle_height: 2.0
          clearing: True
          marking: True
          data_type: "LaserScan"
          raytrace_max_range: 3.0
          raytrace_min_range: 0.0
          obstacle_max_range: 2.5
          obstacle_min_range: 0.0
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 0.55
      stvl_layer:
        plugin: "spatio_temporal_voxel_layer/SpatioTemporalVoxelLayer"
        enabled: true
        voxel_decay: 15.
        decay_model: 0
        voxel_size: 0.05
        track_unknown_space: true
        unknown_threshold: 15
        mark_threshold: 0
        update_footprint_enabled: true
        combination_method: 1
        origin_z: 0.0
        publish_voxel_map: true
        transform_tolerance: 0.2
        mapping_mode: false
        map_save_duration: 60.0
        observation_sources: pointcloud
        pointcloud:
          data_type: PointCloud2
          topic: /intel_realsense_r200_depth/points
          marking: true
          clearing: true
          obstacle_range: 3.0
          min_obstacle_height: 0.0
          max_obstacle_height: 2.0
          expected_update_rate: 0.0
          observation_persistence: 0.0
          inf_is_valid: false
          filter: "voxel"
          voxel_min_points: 0
          clear_after_reading: true
          max_z: 7.0
          min_z: 0.1
          vertical_fov_angle: 0.8745
          horizontal_fov_angle: 1.048
          decay_acceleration: 15.0
          model_type: 0
      always_send_full_costmap: True

map_server:
  ros__parameters:
    use_sim_time: True
    # Overridden in launch by the "map" launch configuration or provided default value.
    # To use in yaml, remove the default "map" value in the tb3_simulation_launch.py file & provide full path to map below.
    yaml_filename: ""

map_saver:
  ros__parameters:
    use_sim_time: True
    save_map_timeout: 5.0
    free_thresh_default: 0.25
    occupied_thresh_default: 0.65
    map_subscribe_transient_local: True

planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    use_sim_time: True
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_navfn_planner/NavfnPlanner"
      tolerance: 0.5
      use_astar: false
      allow_unknown: true

smoother_server:
  ros__parameters:
    use_sim_time: True
    smoother_plugins: ["simple_smoother"]
    simple_smoother:
      plugin: "nav2_smoother::SimpleSmoother"
      tolerance: 1.0e-10
      max_its: 1000
      do_refinement: True

behavior_server:
  ros__parameters:
    costmap_topic: local_costmap/costmap_raw
    footprint_topic: local_costmap/published_footprint
    cycle_frequency: 10.0
    behavior_plugins: ["spin", "backup", "drive_on_heading", "assisted_teleop", "wait"]
    spin:
      plugin: "nav2_behaviors/Spin"
    backup:
      plugin: "nav2_behaviors/BackUp"
    drive_on_heading:
      plugin: "nav2_behaviors/DriveOnHeading"
    wait:
      plugin: "nav2_behaviors/Wait"
    assisted_teleop:
      plugin: "nav2_behaviors/AssistedTeleop"
    global_frame: odom
    robot_base_frame: base_link
    transform_tolerance: 0.1
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    use_sim_time: True
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

velocity_smoother:
  ros__parameters:
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0

```

#### Expected behavior
I can compile with colcon build without problem, and I launch the simulation with
```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False
```
Rviz and Gazebo starts without problem, so I try to set the initial position using the RViz tool to set the 2D position (in the 3° I use `ros2 launch slam_toolbox online_async_launch.py`)

#### Actual behavior
In both cases I get this errors:
```
[component_container_isolated-6] [INFO] [1700652858.760815286] [global_costmap.global_costmap]: start
[component_container_isolated-6] [INFO] [1700652858.761636798] [global_costmap.global_costmap]: stvl_layer was activated.
[rviz2-5] [INFO] [1700652859.882882227] [rviz2]: Trying to create a map of size 384 x 384 using 1 swatches
[component_container_isolated-6] [INFO] [1700652859.913019570] [planner_server]: Activating plugin GridBased of type NavfnPlanner
[component_container_isolated-6] [INFO] [1700652859.914863086] [planner_server]: Creating bond (planner_server) to lifecycle manager.
[component_container_isolated-6] [INFO] [1700652860.021056606] [lifecycle_manager_navigation]: Server planner_server connected with bond.
[component_container_isolated-6] [INFO] [1700652860.021124081] [lifecycle_manager_navigation]: Activating behavior_server
[component_container_isolated-6] [INFO] [1700652860.021420955] [behavior_server]: Activating
[component_container_isolated-6] [INFO] [1700652860.021448282] [behavior_server]: Activating spin
[component_container_isolated-6] [INFO] [1700652860.021472907] [behavior_server]: Activating backup
[component_container_isolated-6] [INFO] [1700652860.021487097] [behavior_server]: Activating drive_on_heading
[component_container_isolated-6] [INFO] [1700652860.021504167] [behavior_server]: Activating assisted_teleop
[component_container_isolated-6] [INFO] [1700652860.021515689] [behavior_server]: Activating wait
[component_container_isolated-6] [INFO] [1700652860.021529797] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[component_container_isolated-6] [INFO] [1700652860.125669477] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[component_container_isolated-6] [INFO] [1700652860.125704458] [lifecycle_manager_navigation]: Activating bt_navigator
[component_container_isolated-6] [INFO] [1700652860.125952765] [bt_navigator]: Activating
[component_container_isolated-6] [ERROR] [1700652861.143982459] [bt_navigator_navigate_through_poses_rclcpp_node]: "ClearLocalCostmap-Context" service server not available after waiting for 1 s
[component_container_isolated-6] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[ERROR] [component_container_isolated-6]: process has died [pid 98861, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_543eu3zk --params-file /tmp/launch_params_66wyy19r -r /tf:=tf -r /tf_static:=tf_static'].
```
Can you please help me to solve this problem? I can not find any solution.
This is my question on [Robotics StackExchange](https://robotics.stackexchange.com/questions/105589/nav2-general-tutorial-problem-on-ros-2-humble).
Thanks in advance.

SteveMacenski, 2023-11-22:
Do not cross post with robotics stack exchange. We'll follow up there

closed 2023-11-22 by SteveMacenski

azazdeaz, 2025-04-01:
I can't find this post on Robotics StackExchange. The link returns 404. Did you find a solution?