ros-navigation/navigation2 #3983: gcc version for build from source

elgarbe, 2023-11-23:
Hi, I need to know what is the gcc version that is needed in order to build from source.
I get some errors building on a docker image and think that is related to gcc installed on the base image I'm using.

Thank

SteveMacenski, 2023-11-23:
We do not specifically change the default GCC version from the host operating system on Tier 1 supported OS’ of your distribution. 

Since you did not include the issue template I can’t tell you what version that is, but you can look it up on the distribution’s package manager.

closed 2023-11-23 by SteveMacenski

elgarbe, 2023-11-23:
Hi, I have this error:

```
[Processing: nav2_behavior_tree, nav2_costmap_2d, ompl, rviz_ogre_vendor]
[ 28%] Building CXX object CMakeFiles/layers.dir/plugins/denoise_layer.cpp.o
[ 48%] Building CXX object src/ompl/CMakeFiles/ompl.dir/multilevel/datastructures/projections/src/None.cpp.o
[ 43%] Linking CXX shared library libnav2_planner_selector_bt_node.so
[ 39%] Building CXX object OgreMain/CMakeFiles/OgreMain.dir/src/OgreLog.cpp.o
[ 43%] Built target nav2_planner_selector_bt_node
[ 44%] Building CXX object CMakeFiles/nav2_controller_selector_bt_node.dir/plugins/action/controller_selector_node.cpp.o
[ 48%] Building CXX object src/ompl/CMakeFiles/ompl.dir/multilevel/datastructures/projections/src/RNSO2_RN.cpp.o
In file included from /usr/include/c++/7/bits/stl_algobase.h:64:0,
                 from /usr/include/c++/7/bits/char_traits.h:39,
                 from /usr/include/c++/7/string:40,
                 from /root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/layer.hpp:40,
                 from /root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise_layer.hpp:18,
                 from /root/nav2_ws/src/navigation2/nav2_costmap_2d/plugins/denoise_layer.cpp:15:
/usr/include/c++/7/bits/stl_pair.h: In instantiation of ‘struct std::pair<nav2_costmap_2d::Image<short unsigned int>, short unsigned int>’:
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:989:56:   required from ‘void nav2_costmap_2d::imgproc_impl::GroupsRemover::removeGroupsImpl(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, nav2_costmap_2d::imgproc_impl::EquivalenceLabelTrees<Label>&, size_t, const IsBg&) const [with nav2_costmap_2d::ConnectivityType connectivity = (nav2_costmap_2d::ConnectivityType)4; Label = short unsigned int; IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:972:37:   required from ‘bool nav2_costmap_2d::imgproc_impl::GroupsRemover::tryRemoveGroupsWithLabelType(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, size_t, nav2_costmap_2d::imgproc_impl::EquivalenceLabelTrees<Label>&, const IsBg&, bool) const [with nav2_costmap_2d::ConnectivityType connectivity = (nav2_costmap_2d::ConnectivityType)4; Label = short unsigned int; IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:936:59:   required from ‘void nav2_costmap_2d::imgproc_impl::GroupsRemover::removeGroupsPickLabelType(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, size_t, const IsBg&) const [with nav2_costmap_2d::ConnectivityType connectivity = (nav2_costmap_2d::ConnectivityType)4; IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:908:7:   required from ‘void nav2_costmap_2d::imgproc_impl::GroupsRemover::removeGroups(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, nav2_costmap_2d::ConnectivityType, size_t, const IsBg&) const [with IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
/root/nav2_ws/src/navigation2/nav2_costmap_2d/plugins/denoise_layer.cpp:156:56:   required from here
/usr/include/c++/7/bits/stl_pair.h:303:17: error: ‘constexpr std::pair<_T1, _T2>::pair(const std::pair<_T1, _T2>&) [with _T1 = nav2_costmap_2d::Image<short unsigned int>; _T2 = short unsigned int]’ declared to take const reference, but implicit declaration would take non-const
       constexpr pair(const pair&) = default;
                 ^~~~
/usr/include/c++/7/bits/stl_pair.h: In instantiation of ‘struct std::pair<nav2_costmap_2d::Image<unsigned int>, unsigned int>’:
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:989:56:   required from ‘void nav2_costmap_2d::imgproc_impl::GroupsRemover::removeGroupsImpl(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, nav2_costmap_2d::imgproc_impl::EquivalenceLabelTrees<Label>&, size_t, const IsBg&) const [with nav2_costmap_2d::ConnectivityType connectivity = (nav2_costmap_2d::ConnectivityType)4; Label = unsigned int; IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
/root/nav2_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/denoise/image_processing.hpp:972:37:   required from ‘bool nav2_costmap_2d::imgproc_impl::GroupsRemover::tryRemoveGroupsWithLabelType(nav2_costmap_2d::Image<unsigned char>&, nav2_costmap_2d::MemoryBuffer&, size_t, nav2_costmap_2d::imgproc_impl::EquivalenceLabelTrees<Label>&, const IsBg&, bool) const [with nav2_costmap_2d::ConnectivityType connectivity = (nav2_costmap_2d::ConnectivityType)4; Label = unsigned int; IsBg = nav2_costmap_2d::DenoiseLayer::removeGroups(nav2_costmap_2d::Image<unsigned char>&) const::<lambda(uint8_t)>; size_t = long unsigned int]’
[... cut ...]
     transform(_InputIterator __first, _InputIterator __last,
     ^~~~~~~~~
cc1plus: all warnings being treated as errors
CMakeFiles/layers.dir/build.make:159: recipe for target 'CMakeFiles/layers.dir/plugins/denoise_layer.cpp.o' failed
make[2]: *** [CMakeFiles/layers.dir/plugins/denoise_layer.cpp.o] Error 1
CMakeFiles/Makefile2:294: recipe for target 'CMakeFiles/layers.dir/all' failed
Makefile:145: recipe for target 'all' failed
make[1]: *** [CMakeFiles/layers.dir/all] Error 2
make: *** [all] Error 2
--- stderr: nav2_costmap_2d

```