ros-navigation/navigation2 #3984: gcc version for build from source

elgarbe, 2023-11-23:
We do not specifically change the default GCC version from the host operating system on Tier 1 supported OS’ of your distribution. 

Since you did not include the issue template I can’t tell you what version that is, but you can look it up on the distribution’s package manager.

_Originally posted by @SteveMacenski in https://github.com/ros-planning/navigation2/issues/3983#issuecomment-1823925981_

I get this error:

SteveMacenski, 2023-11-23:
Please don't file new tickets. Comment on existing ones. 

You still did not provide the issue template context, so I cannot help. Please direct this to Robotics Stack Exchange in the future or look into the errors.

closed 2023-11-23 by SteveMacenski

elgarbe, 2023-11-23:
Usually I file a ticket on github project and after a couple of posts we decide to close the issue as solved or not. I don't know why yuo close the issue and don't let me answer yuor requests.
Anyway:

I'm working on building Nav2 from source on a Dockerfile for the Jetson Nano. The official ubuntu distro is 18.04 (no ROS2 for it), so I'm using https://github.com/dusty-nv/jetson-containers as a base image. My Dockerfile is:

```
FROM dustynv/ros:humble-ros-base-l4t-r32.7.1

RUN mkdir -p /root/nav2_ws/src
WORKDIR /root/nav2_ws

ENV ROS_PACKAGE_PATH=/opt/ros/humble/install

# ROS fundamentals
RUN apt-mark hold '*opencv*' \
    && apt-get update \
    && apt autoremove -y \
    && rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall \
    && vcs import src/ < ros2.humble.nav2.rosinstall \
    && . /opt/ros/$ROS_DISTRO/install/setup.sh \
    && rosdep fix-permissions \
    && rosdep update \
    && rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"

RUN echo "Fixing client.hpp file" \
    && sed -i "s|return value|return std::move(value) |g" /opt/ros/humble/install/include/rclcpp/rclcpp/client.hpp \
    && echo "client.hpp fixed"  

# ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp
# COPY ros_entrypoint.sh /
# ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]

```
- Operating System: Ubuntu 22.04 in docker container
- ROS2 Version: Humble building from source on the base image
- Version or commit hash: fc2a199a4c1f8d8cfd8abdb216685b47508e01b2

#### Steps to reproduce issue

Create a Dockerfile with provided code and run 
```
docker build -t nav2_docker -f Dockerfile .
docker run -it nav2_docker:latest

```
Once incide de docker container run
`colcon build --symlink-install --base-paths src --event-handlers console_direct+`
or
`colcon build --symlink-install --base-paths src --event-handlers console_direct+ &> results.txt`

#### Expected behavior
The build process finishd without errors

#### Actual behavior
The build process finishd with reported error

#### Additional information
I think that the problem is related to the version of the gcc that comes with dustynv/ros:humble-ros-base-l4t-r32.7.1, maybe it's too old and doesn't recognize the code that generate the error. But I'm noob at c++.

Thank