ros-navigation/navigation2 #3996: In `bt_service_node.hpp`, `bt_loop_duration` is not in the ProvidedPorts list

AndyZe, 2023-11-30:
I think this is an oversight b/c, for example, `server_timeout` is included in `providedBasicPorts()`.

https://github.com/ros-planning/navigation2/blob/c1e4a427b120da9aea40b78d34e71d68a53c5379/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L113C49-L113C63

SteveMacenski, 2023-11-30:
These are indeed different parameters for different things and `bt_loop_duration` is not a parameter that is adjustable on a per-BT node basis. Its a system level parameter of the tick rate of the entire tree. Its used to limit the action/service call timeouts to the BT tick rate so that the tree can obtain the full BT tick rate requested by the user. So, this is not changable beyond adjusting that parameter for the whole tree.

The server timeout is the total time to wait for a result from an action / service that can be over the course of multiple BT node ticks that were preempted by the BT loop duration's increments

AndyZe, 2023-11-30:
OK, thanks. Not completely clear from looking at `bt_service_node.hpp` but I can't think of a way to make it better.

closed 2023-11-30 by AndyZe