ros-navigation/navigation2 #4001: Bug in smoothPath Function of Simple Commander API on Humble Branch

brian7989, 2023-12-06:
I've identified a bug in the nav2_simple_commander package, specifically within the smoothPath function. This issue seems to have been introduced in commit **0d86469f1c31708117e68f3f44ad5392347e9a5d** on the **Humble** branch.

### Description of the Issue:

The `smoothPath` function in the BasicNavigator class is incorrectly passing an extra self argument to the `_smoothPathImpl` method. This results in a TypeError as `_smoothPathImpl` receives one more argument than it expects.

### Location of the Bug:
The issue is located in the smoothPath method definition:

```
def smoothPath(self, path, smoother_id='', max_duration=2.0, check_for_collision=False):
    """Send a `SmoothPath` action request."""
    rtn = self._smoothPathImpl(
        self, path, smoother_id, max_duration, check_for_collision)
```

### Suggested Fix:
The extra self argument should be removed from the call to _smoothPathImpl. The corrected function call should be:

`rtn = self._smoothPathImpl(path, smoother_id, max_duration, check_for_collision)`

### Impact:
This bug prevents the proper use of the smoothPath function, which is a critical feature for many applications using the nav2_simple_commander package for navigation tasks.

SteveMacenski, 2023-12-06:
@brian7989 can you open that PR?

SteveMacenski, 2024-01-03:
Its only in Humble actually - doing myself

commit d29de3db (2024-01-03): Fix #4001 (#4028)
closed 2024-01-03 by SteveMacenski