ros-navigation/navigation2 #4003: BT XML Idea: Stop replanning when near the goal if path is valid

SteveMacenski, 2023-12-07:
When using feasible planners, it might not be sensible to replan when things are clear within X meters of the goal. It can actually be detrimental to do so if the finite change in path tracking error makes exact goals not possible once within < 1m to the goal result in some "looping" behaviors. This is totally unnecessary to achieve the goal successfully given the positional and rotational tolerances that the exact planners don't know about.

So, it seems very reasonable to use the goal distance BT node to check if we're with `x` meters of the goal and if so, select not to replan if the path continues to be totally valid (e.g. no occopancy). This could be added as either a new BT or modify existing BTs to take advantage of this knowledge

VArt3m, 2024-01-26:
That's a very sensible thing to do. Was thinking about doing something like that to prevent loops when near goal myself - it may be possible to use `GoalReached` BT node with high (1-3m) threshold for that purpose. Although I suspect that having a "path length" condition would be more correct - but there's no node for that yet.

P.S. I find it a bit strange that `GoalReached` BT node *doesn't* leverage goal checker plugins from Controller

SteveMacenski, 2024-01-30:
@shannonwarren Lets discuss here:

Making it based on a timer seems excessive, we could simply add a control flow node under the current rate timer which uses a condition node [`isGoalReached`](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/condition/goal_reached_condition.cpp) -- or more accurately would be to make a new one using the `nav2_util` for getting the integrated path distance. 

If we're within the distance to the goal spatially or by integrated path length, then we can use [`isPathValid`](https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/condition/is_path_valid_condition.cpp) to check if the path is collision free. If its in collision, then replan. If not, pass on replanning.

We should be able to do this with no changes to existing BT nodes and possibly just creating 1 new BT node which just wraps an existing `nav2_util` function. Then its just organizing those stages using the basic control flow nodes

shannonwarren, 2024-01-31:
Sounds good. I'll work on it.

anath93, 2024-02-14:
@SteveMacenski @shannonwarren I would like to help being first time contributor to the Nav2 community, and learn along the side. I wont be able to PR it, as currently I am on Galactic and have rebased planners and util from main branch, but happy to draft everything up here if someone wants to publish it to the repo, after approval.
So far this is what I have .hpp and .cpp for condition node if robot is within reach of goal using integrated path from Nav2 util side, please let me know your thoughts and mistakes if any or any changes to naming convention.

goal_reached_replanning_condition.hpp

`// Copyright (c) 2019 Intel Corporation
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#ifndef NAV2_BEHAVIOR_TREE__PLUGINS__CONDITION__GOAL_REACHED_REPLANNING_CONDITION_HPP_
#define NAV2_BEHAVIOR_TREE__PLUGINS__CONDITION__GOAL_REACHED_REPLANNING_CONDITION_HPP_

#include <string>
#include <memory>
#include <limits>

#include "rclcpp/rclcpp.hpp"
#include "nav_msgs/msg/path.hpp"
#include "behaviortree_cpp_v3/condition_node.h"
#include "geometry_msgs/msg/pose_stamped.hpp"

namespace nav2_behavior_tree
{

/**
 * @brief A BT::ConditionNode that returns SUCCESS when a specified goal
 * is within tolerance by making use of integrated path distance.
 */
class GoalReachedReplanningCondition : public BT::ConditionNode
{
public:
  /**
   * @brief A constructor for nav2_behavior_tree::GoalReachedReplanningCondition
   * @param condition_name Name for the XML tag for this node
   * @param conf BT node configuration
   */
  GoalReachedReplanningCondition(
    const std::string & condition_name,
    const BT::NodeConfiguration & conf);

  GoalReachedReplanningCondition() = delete;

  /**
   * @brief A destructor for nav2_behavior_tree::GoalReachedReplanningCondition
   */
  ~GoalReachedReplanningCondition() override;

  /**
   * @brief The main override required by a BT action
   * @return BT::NodeStatus Status of tick execution
   */
  BT::NodeStatus tick() override;

  /**
   * @brief Function to read parameters and initialize class variables
   */
  void initialize();

  /**
   * @brief Checks if the current robot pose lies within a given distance from the goal
   * @return bool true when goal is reached, false otherwise
   */
  bool isRobotWithinGoal();

  /**
   * @brief Creates list of BT ports
   * @return BT::PortsList Containing node-specific ports
   */
  static BT::PortsList providedPorts()
  {
    return {
      BT::InputPort<nav_msgs::msg::Path>("path", "Planned Path"),
      BT::InputPort<double>("len_check", 1.0,"Length(m) before goal for which Robot will not replan"),
    };
  }

protected:
  /**
   * @brief Cleanup function
   */
  void cleanup()
  {}

private:
  nav_msgs::msg::Path path_;
  double len_check_ = std::numeric_limits<double>::max();
  rclcpp::Node::SharedPtr node_;
  bool initialized_;
};

}  // namespace nav2_behavior_tree

#endif  // NAV2_BEHAVIOR_TREE__PLUGINS__CONDITION__GOAL_REACHED_REPLANNING_CONDITION_HPP_`

goal_reached_replanning_condition.cpp

`// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <string>
#include <memory>
#include <vector>
#include "nav2_util/geometry_utils.hpp"

#include "nav2_behavior_tree/plugins/condition/goal_reached_replanning_condition.hpp"

namespace nav2_behavior_tree
{

GoalReachedReplanningCondition::GoalReachedReplanningCondition(
  const std::string & condition_name,
  const BT::NodeConfiguration & conf)
: BT::ConditionNode(condition_name, conf),
  initialized_(false),
  len_check_(1.0)
{
  node_ = config().blackboard->get<rclcpp::Node::SharedPtr>("node");
}

GoalReachedReplanningCondition::~GoalReachedReplanningCondition()
{
  cleanup();
}

BT::NodeStatus GoalReachedReplanningCondition::tick()
{
  if (!initialized_) {
    initialize();
  }

  if (isRobotWithinGoal()) {
    return BT::NodeStatus::SUCCESS;
  }
  return BT::NodeStatus::FAILURE;
}

void GoalReachedReplanningCondition::initialize()
{
  node_ = config().blackboard->get<rclcpp::Node::SharedPtr>("node");
  node_->get_parameter_or<double>("len_check", len_check_, 1.0);
  initialized_ = true;
}

bool GoalReachedReplanningCondition::isRobotWithinGoal()
{
  nav_msgs::msg::Path path;
  double len_check;
  double dist_to_goal = nav2_util::geometry_utils::calculate_path_length(path,find_closest_pose_idx());

  if (dist_to_goal <= len_check && dist_to_goal >= 0.0)
  {
    RCLCPP_DEBUG(node_->get_logger(), "Robot in Goal Proximity.");
    return true;
  }

  return false;
}

}  // namespace nav2_behavior_tree

#include "behaviortree_cpp_v3/bt_factory.h"
BT_REGISTER_NODES(factory)
{
  factory.registerNodeType<nav2_behavior_tree::GoalReachedReplanningCondition>("GoalReachedReplanningCondition");
}
`

padhupradheep, 2024-04-11:
Did something similar for another project.. 

however, when the robot is in the goal radius with an obstacle, I just wait and not go for replanning.. 

https://github.com/ros-planning/navigation2/assets/20242192/60a72ab1-0507-4162-b836-2e836dddf428

SteveMacenski, 2025-07-24:
We have some designs for that in Nav2 in various places (though not in all... which is something we should fix). For cases in which we call the function regularly (i.e. controller server at a fixed frequency, not planner server since that could be called infrequently or irregularly), we can:

* Use the first path point to start
* Have a parameter for forward search distance to check, which can be set as maximum speed * update rate + some healthy margin (or 1/2 costmap size as a good default). This is set as the maximum distance the robot can possibly move in an iteration. 
* Check from the start -> that distance for the nearest point, only
* Return the value
* Next iteration start with this point to seed the problem. This could be done by storing and setting the last-closest IDX on the path, or by pruning the path behind that pose at the end of the method so that we initialize each at the first point of the path. 

This seems to work well. Open to other ideas, but this is what is 'proven' at least, for the case in which we update regularly to know what a maximum forward distance should be to check. If we don't know that, we may still need to check the full path, but maybe have a prune-passed-points feature to remove former loops from the equation. Deciding on future loops is a little more involved, also considering paths that may be near each other and not necessarily looping (like U turns)

mini-1235, 2025-07-24:
I think this is also related to the distance_remaining feedback in NavToPose and NavThroughPose https://github.com/ros-navigation/navigation2/blob/332bcab9b51cc2d0472e554132590c7bc6974baf/nav2_bt_navigator/src/navigators/navigate_through_poses.cpp#L153-L174
Do you think they both need to be updated?

SteveMacenski, 2025-07-25:
Ideally speaking, yes. If we can actually abstract this process into a library in nav2_utils that would be _ideal_ so we can just use it across the board more easily rather than redoing it in multiple places. The API might differ a little bit in a few places, I'm not sure the full list of locations where we try to do something like this. I'm guessing at least in the controller's path handlers / handling processes and in the BT Navigator's feedback. The Navigators are another good place for the procedure I outlined since that also runs at a regular frequency. 

I don't think Nav To Pose needs it as much, but I figure it doesn't hurt if we apply it to both Navigators if they share an implementation in a utils file.

mini-1235, 2026-01-14:
Should we close this as https://github.com/ros-navigation/navigation2/pull/5843 is merged?

SteveMacenski, 2026-01-14:
I think this is to actually have a BT (perhaps as part of the default BT updates sprint) that uses this, not just that it should be possible to be able to. I think this is really sane default behavior to have // behavior that is demonstrated for those looking for ideas