ros-navigation/navigation2 #4007: Create arc-shaped path

MinElite, 2023-12-12:
Dear all,

I've been working on creating an arc-shaped path through a new planner plugin. Unlike the usual "normal" paths with a start and end pose, an arc path requires a third point for determining its center and radius. Following the tutorial, I noticed that the createPlan() method supports only two inputs, which isn't sufficient for my needs.

I was wondering if it's possible to add a new virtual method, say createPlanExt(), that accepts a third input. I attempted to define it in nav_core as a pure virtual function, but did not work well. Any guidance on this or suggestions for a better approach would be greatly appreciated!

Cheers,
Min

SteveMacenski, 2023-12-12:
The API is built for generic path planners, if you have a very bespoke application doing something unusual, you'd have to create your own server to meet your API needs. An example is [Open Navigation's coverage server](https://github.com/open-navigation/opennav_coverage) who does complete-coverage paths. 

I will say that from a distance without much specific knowledge of your application or use-case, this seems more like a toy example than a production utility. I wouldn't personally bother with trying to go to that level of effort and instead try to use the existing API with parameters that specify these features instead of modifying the API or trying to build your own server for a sample application instead of something you truly need. 

We won't though just modify the API to take in an arbitrary 3rd input that has no specific meaning or application which can be interpreted freely without bounds.

closed 2023-12-12 by SteveMacenski

MinElite, 2023-12-13:
Thanks a bunch for the speedy reply.
We're currently working on a fleet management demo, focusing on task allocation for multiple mobile robots. Each robot needs to follow specific pre-specified paths consist of various path primitives like straight lines, arcs, Bezier curves, etc. I totally get that modifying the existing nav2 APIs might not be the best route. After some thought, I'm considering an alternative approach. I'm thinking of writing a separate script for path generation and then using commander APIs such as followPath() or goThroughPoses() for navigation. I hope this sounds like a reasonable plan. 
Thanks again, and wishing you a nice day!

SteveMacenski, 2023-12-13:
That seems like potentially a better solution, yes. This API is for free-space planning. 

Though, by example, the State Lattice planner uses a minimum control set (a set of primitives) to plan routes. But that’s still dynamically searching and not fully pre-computed. If you have precomputed routes and don’t require dynamic freespace updates, then I don’t think this is the right place to modify. If you have a system in place you prefer, then you can call it in the BT XML in place of ComputePathToPose and the rest should work as expected 🙂