ros-navigation/navigation2 #4010: nav2_mppi_controller colcon build error

THARUN-V, 2023-12-13:
## Bug report

**Required Info:**

- Operating System:
  - UBUNTU 22
- ROS2 Version:
  - HUMBLE
[image]

SteveMacenski, 2023-12-13:
Builds fine on my machine, many other users, and in CI. What's your OS and build settings? Anything unusual or custom? This is likely something on your side

THARUN-V, 2023-12-13:
i tried in two machines running ubuntu 22 and humble , same issue persists.

SteveMacenski, 2023-12-13:
I bootstrapped a new machine last week, I just installed Humble on it to test this, pulled auto dependencies from rosdep, and built it fine. 

```
Starting >>> nav2_common
Finished <<< nav2_common [0.38s]                  
Starting >>> nav2_msgs
Starting >>> nav2_voxel_grid            
Finished <<< nav2_voxel_grid [18.0s]                                                        
Finished <<< nav2_msgs [26.3s]                       
Starting >>> nav2_util
[Processing: nav2_util]                             
Finished <<< nav2_util [30.4s]                       
Starting >>> nav2_lifecycle_manager
Starting >>> nav2_map_server
Finished <<< nav2_lifecycle_manager [26.0s]                                                                
Finished <<< nav2_map_server [38.9s]                            
Starting >>> nav2_costmap_2d
[Processing: nav2_costmap_2d]                                  
[Processing: nav2_costmap_2d]                                      
[Processing: nav2_costmap_2d]                                       
Finished <<< nav2_costmap_2d [1min 45s]                              
Starting >>> nav2_core
Finished <<< nav2_core [2.59s]                        
Starting >>> nav2_mppi_controller
[Processing: nav2_mppi_controller]                              
[Processing: nav2_mppi_controller]                                      
[Processing: nav2_mppi_controller]                                       
[Processing: nav2_mppi_controller]                                       
Finished <<< nav2_mppi_controller [2min 8s]                              

Summary: 9 packages finished [5min 32s]
```

The binaries are also turning over in the CI build farm, which is ground truth for everything https://build.ros2.org/job/Hbin_uJ64__nav2_mppi_controller__ubuntu_jammy_amd64__binary/ & in our own internal CI systems. 

Not sure what else to tell you - I believe this has to do with your system and since I can't reproduce it locally in docker, bare metal, nor remotely on build farm & CI, nor am I hearing anything from any other users (which there are dozens using MPPI on Humble), the only logical result I can come up with is it has to do with configurations on your side :upside_down_face: 

With this amount of information, I can't provide much help. You'd be best spent trying to figure out what your system is doing irregularly from standard or what flags you've set that aren't default. :frowning: 

Sorry that I can't be more help on this!

closed 2023-12-13 by SteveMacenski