ros-navigation/navigation2 #4014: set size_x_ and size_y_ in method initMaps

Hao-Li-Bachelorarbeit, 2023-12-17:
## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

in file costmap_2d.cpp
```
void Costmap2D::initMaps(unsigned int size_x, unsigned int size_y)
{
  std::unique_lock<mutex_t> lock(*access_);
  delete[] costmap_;
  costmap_ = new unsigned char[size_x * size_y];
}
```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

set size_x_ and size_y_ in this method. Since these two attributes are always used to describe costmap_, I think this information should be synchronized when initializing the map, and setting them separately causes code redundancy.
#### Implementation considerations

SteveMacenski, 2023-12-17:
All uses of it now either call it using the member sizes or set the new member sizes right before the call (e.g. resizeMap). You could do the same in your application of just setting those member sizes before you call `initMaps`. But, I have no objection to this, can you submit the PR?

Hao-Li-Bachelorarbeit, 2023-12-18:
Already submitted a PR

SteveMacenski, 2023-12-18:
I requested a quick change, otherwise LGTM

Hao-Li-Bachelorarbeit, 2023-12-18:
Done, please notice the PR

closed 2023-12-18 by SteveMacenski