ros-navigation/navigation2 #4016: [humble binary] BT: „missing port [value]” for blackboard “{goal}”

lmendyk, 2023-12-18:
I’m writing my first BT in which I want to reuse the default one for navigate to pose in a way to make a robot go first to   the goal and after that goes back so I can use the GUI for navigation the turtlebot3.  

To do that I’m using the subtree which definition is from the default BT “navigate_to_pose_w_replanning_and_recovery.xml” but used as a subtree.  In the main tree  in the sequence I intend to call the subtree which uses {goal} (and {start} variables and next I want to call the same tree but with {goal} and {start} swapped to make the robot come back.

First I wanted to use the re-mapping mechanism, but from my understanding there are bugs which fixes may not have been released yet ( https://github.com/ros-planning/navigation2/issues/3640  and https://github.com/BehaviorTree/BehaviorTree.CPP/issues/563) so I decided to use the “workaround” and use the mechanism with   __shared_blackboard="true" but for this to work I need to save the {goal} (and {start}  variables.  
I try doing  it with that:
`<SetBlackboard value="{goal}" output_key="orignalGoal" />`

But I'm getting an error: 
> Behavior tree threw exception: missing port [value]. Exiting with failure

My understating of this error is that {goal}  must be  returning Null or similar causing that error.   But As I understand the Nav stack wrapping BT should have defined the {goal} key value in the blackboard as 
`<ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/> `
Does not cause the problem.

Could you advice what is wrong with the idea? 
The snippet of my BT is as follows (there are commented lines – the effect of my debugging :

```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <Sequence>
      <!--<Script code=" orignalStart:=start; orignalGoal:=goal; "/>-->
      <!--<SetBlackboard value="{start}" output_key="orignalStart" />-->
      <!--<SetBlackboard value="{goal}" output_key="orignalGoal" />-->
      <!--<SubTree ID="NavigateToPoseTree" goal="{orignalGoal}" node="{node}" bt_loop_duration="{bt_loop_duration}" server_timeout="{server_timeout}"/> -->
      <!-- <SubTree ID="NavigateToPoseTree" goal="{orignalStart}" node="{node}" bt_loop_duration="{bt_loop_duration}" server_timeout="{server_timeout}"/> -->
      <SubTree ID="NavigateToPoseTree" __shared_blackboard="true"/>
      <SetBlackboard value="{goal}" output_key="orignalGoal" />   <!-- when commented out - there is no error -->
      <!--<SetBlackboard value="{orignalGoal}" output_key="start" /> -->
      <!--<SetBlackboard value="{orignalStart}" output_key="goal" /> -->
      <SubTree ID="NavigateToPoseTree" __shared_blackboard="true" /> 

    </Sequence>
  </BehaviorTree>

```

SteveMacenski, 2023-12-18:
Asking RSE, this is not the right place to obtain general behavior tree Q&A.

closed 2023-12-18 by SteveMacenski

lmendyk, 2023-12-18:
I would consider it it is a bug and probably tricky one BT vs Nav2 wrapper, not just generic question on how BT behaves as I don't see the reason it should not work correctly.

SteveMacenski, 2023-12-18:
That is not a question for Nav2, but BT.CPP. This is not the right place for generic help - please read BT.CPP documentation and if you still have any questions, ask on RSE.

lmendyk, 2023-12-18:
I think it is a bug between Nav2 and BT.CPP  (BT.CPP may not know anything how {goal} is defined by Nav2.)  My suspicion is that {goal} may not serialize to String).    

Not quite sure what you mean asking on RSE - as I'm reporting a bug not just asking the question.

facontidavide, 2023-12-19:
FYI, I just wrote a unit test and both version 3.8 and 4.4 pass without any issue

lmendyk, 2023-12-19:
Thank you for prompt response.

I’m doing it as I hoped it could be a workaround for problem with port mapping when using subtree (I provided the ref to issue in Nav2 stack which provides bigger context  https://github.com/ros-planning/navigation2/issues/4016).

I’m just guessing that this is because the {goal} is not a String value but geometry_msgs::msg::PoseStamped so that may be the reason it does not work. I reported it initially in Nav2 as I suspected that this issue is between Nav2-BT. 

I got enthusiasm about BT as a tool for expressing complex logic using the concept of low-code/no-code so I may not be able to provide you the c++ test case as programing in c++ is sth which I wanted to avoid and use just XML defined BT.

But if SetBlackboard was not meant for copying a value from another key entry, than  need to learn how to build Nav2 with BT from sources to have   the bugs related to subtree ports remapping fixed (https://github.com/ros-planning/navigation2/issues/3640 )

Thank you anyway for quick response.