ros-navigation/navigation2 #4024: SimpleActionServer may call `nullprt` callback when executing the transition to deactivate

BriceRenaudeau, 2023-12-21:
## Bug report

The SimpleActionServer may call `nullprt` callback when executing the transition to deactivate if it is running and no completion callback is set.

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - ROS2 Iron
- Version or commit hash:
  - Latest Iron branch
- DDS implementation:
  - cycloneDDS

#### Steps to reproduce issue

Call lifecycle service transition "deactivate" when a lifecycle node with a simple action server is running. 

#### Expected behavior

The SimpleActionServer doesn't call any callback.

#### Actual behavior

The SimpleActionServer does call a `nullprt` callback creating `bad_function_call` exception and the transition fails.

#### Additional information

The transition function catches a `bad_function_call` exception.
The `deactivate()` method will set `stop_execution_ = true;
https://github.com/ros-planning/navigation2/blob/c213c85c110f5f1d2b689c240ce84fb01eb430f5/nav2_util/include/nav2_util/simple_action_server.hpp#L299-L307

In the `work()` method, the code will call the `completion_callback_` callback, but by default, this callback is  `nullptr`.
https://github.com/ros-planning/navigation2/blob/c213c85c110f5f1d2b689c240ce84fb01eb430f5/nav2_util/include/nav2_util/simple_action_server.hpp#L262-L267

https://github.com/ros-planning/navigation2/blob/c213c85c110f5f1d2b689c240ce84fb01eb430f5/nav2_util/include/nav2_util/simple_action_server.hpp#L64-L68

BriceRenaudeau, 2023-12-21:
The easy way to fix this is to check `if(completion_callback_)` before executing it.

SteveMacenski, 2023-12-21:
Fair take, can you submit the PR?

BriceRenaudeau, 2023-12-21:
Yes, I will.

closed 2023-12-21 by SteveMacenski