ros-navigation/navigation2 #4027: navigation not working

sys-met, 2023-12-31:
I'm having navigation issues in my real robot app. Does anyone have any suggestions?

**Required Info:**

- Operating System:
  - ubuntu 20.04
- ROS2 Version:
  - galactic
- Version or commit hash:
  - ros-galactic-nav2-amcl	1.0.12-1focal.20221208.073405
ros-galactic-nav2-behavior-tree	1.0.12-1focal.20221208.073533
ros-galactic-nav2-bringup	1.0.12-1focal.20221208.082839
ros-galactic-nav2-bt-navigator	1.0.12-1focal.20221208.075615
ros-galactic-nav2-common	1.0.12-1focal.20221208.070118
ros-galactic-nav2-controller	1.0.12-1focal.20221208.075637
ros-galactic-nav2-core	1.0.12-1focal.20221208.075333
ros-galactic-nav2-costmap-2d	1.0.12-1focal.20221208.074030
ros-galactic-nav2-dwb-controller	1.0.12-1focal.20221208.081126
ros-galactic-nav2-lifecycle-manager	1.0.12-1focal.20221208.073619
ros-galactic-nav2-map-server	1.0.12-1focal.20221208.073451
ros-galactic-nav2-msgs	1.0.12-1focal.20221208.071323
ros-galactic-nav2-navfn-planner	1.0.12-1focal.20221208.075902
ros-galactic-nav2-planner	1.0.12-1focal.20221208.080258
ros-galactic-nav2-recoveries	1.0.12-1focal.20221208.080258
ros-galactic-nav2-regulated-pure-pursuit-controller	1.0.12-1focal.20221208.080240
ros-galactic-nav2-rotation-shim-controller	1.0.12-1focal.20221208.080642
ros-galactic-nav2-rviz-plugins	1.0.12-1focal.20221208.075041
ros-galactic-nav2-smac-planner	1.0.12-1focal.20221208.080441
ros-galactic-nav2-theta-star-planner	1.0.12-1focal.20221208.080629
ros-galactic-nav2-util	1.0.12-1focal.20221208.072658
ros-galactic-nav2-voxel-grid	1.0.12-1focal.20221208.071338
ros-galactic-nav2-waypoint-follower	1.0.12-1focal.20221208.080633

- DDS implementation:
  - 

#### Steps to reproduce issue

`[INFO] [1704044026.366409294] [controller_server]: Received a goal, begin comput                                                                                                                                                   ing control effort.                                                                                                                                         
[WARN] [1704044026.366921387] [controller_server]: No goal checker was specified                                                                                                                                                    in parameter 'current_goal_checker'. Server will use only plugin loaded general                                                                                                                                                   _goal_checker . This warning will appear once.                                                                                                                                                                                     
[INFO] [1704044027.469682779] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044028.469687944] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044029.569699182] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044030.569678935] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044031.669684518] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044032.669720935] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044033.669683386] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044034.770931629] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044035.769683456] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[ERROR] [1704044036.370203468] [controller_server]: Failed to make progress                                                                                                                                                        
[WARN] [1704044036.370358025] [controller_server_rclcpp_node]: [follow_path] [Ac                                                                                                                                                   tionServer] Aborting handle.                                                                                                                                                                                                       
[Thread 0x7f87795950 (LWP 10832) exited]                                                                                                                                                                                           
[INFO] [1704044036.387111081] [local_costmap.local_costmap]: Received request to                                                                                                                                                    clear entirely the local_costmap                                                                                                                                                                                   
[New Thread 0x7f86f94950 (LWP 10856)]                                                                                                                                                                                              
[INFO] [1704044036.391125286] [controller_server]: Received a goal, begin comput                                                                                                                                                   ing control effort.                                                                                                                                         
[INFO] [1704044036.793030889] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044037.893147911] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[INFO] [1704044038.893015598] [controller_server]: Passing new path to controlle                                                                                                                                                   r.                                                                                      
[... cut ...]
#7  0x0000007fb7e8ff80 in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /opt/ros/galactic/lib/librclcpp.so
#8  0x0000007fb7e902ac in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) ()
   from /opt/ros/galactic/lib/librclcpp.so
#9  0x0000007fb7e941f8 in rclcpp::executors::SingleThreadedExecutor::spin() ()
   from /opt/ros/galactic/lib/librclcpp.so
#10 0x0000007fb7e92c1c in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) () from /opt/ros/galactic/lib/librclcpp.so
`

SteveMacenski, 2024-01-01:
Galactic is EOL and won't receive any support unfortunately. Please upgrade to a currently supported ROS 2 distribution (Humble or newer)

closed 2024-01-01 by SteveMacenski

sys-met, 2024-01-04:
Hi . I raised humble. Navigation failed again. I'm sharing a picture because I can't copy. I tried navigation with slam_toolbox localization and nav2_bringup localization. I got almost the same result

Operating System
  ubuntu 22.04, arm, humble

Version
  ros-humble-navigation2	1.1.12-1jammy.20231206.072740

[image]

SteveMacenski, 2024-01-04:
What is the issue? You signaled a Control+C `^C` so of course it exited.

sys-met, 2024-01-06:
It's similar to the problem I encountered in Galactic.

terminal

> [lifecycle_manager-8] [INFO] [1704534384.853787268] [lifecycle_manager_navigation]: Creating bond timer...
[bt_navigator-5] [INFO] [1704534438.005770732] [bt_navigator]: Begin navigating from current location (-0.02, -0.03) to (1.46, 0.05)
[controller_server-1] [INFO] [1704534438.028356473] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [WARN] [1704534438.028571421] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded general_goal_checker . This warning will appear once.
[controller_server-1] [INFO] [1704534439.079135100] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534440.128998675] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534441.128997328] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534442.179026059] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534443.178967628] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534444.229270057] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534445.229007772] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534446.279020410] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534447.278997031] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1704534448.079202974] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1704534448.080796465] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1704534448.096900536] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-1] [INFO] [1704534448.099923819] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [WARN] [1704534448.172397503] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[controller_server-1] [INFO] [1704534448.351850983] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534449.351139271] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534450.400559356] [controller_server]: Passing new path to controller.
[controller_server-1] [WARN] [1704534450.453034227] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[controller_server-1] [WARN] [1704534451.082163621] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[controller_server-1] [WARN] [1704534451.153570378] [controller_server]: Control loop missed its desired rate of 20.0000Hz
[controller_server-1] [INFO] [1704534451.403667906] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534452.453657939] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534453.454996384] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534454.503655739] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534455.503654080] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534456.553804477] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534457.553648286] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1704534458.203793577] [controller_server]: Failed to make progress
[controller_server-1] [WARN] [1704534458.205626078] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1704534458.226482131] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-3] [INFO] [1704534458.227484788] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-3] [WARN] [1704534458.853917670] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 1.6096 Hz
[controller_server-1] [INFO] [1704534458.870450023] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [INFO] [1704534459.921064431] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534460.920932146] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534461.970926815] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534463.020925130] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534464.020923939] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534465.070923503] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534466.070926166] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534467.120923543] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534468.170923889] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1704534468.871057279] [controller_server]: Failed to make progress
[... cut ...]
[controller_server-1] [INFO] [1704534470.189916204] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534471.239909571] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534472.289904448] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534473.289941173] [controller_server]: Passing new path to controller.
[controller_server-1] [INFO] [1704534474.340138707] [controller_server]: Passing new path to controller.

[image]

sys-met, 2024-01-07:
The problem was that I forgot the cmd_vel setting.  I set it up with twist_mux and it works fine.

SteveMacenski, 2024-01-08:
Glad you found your issue!

WaelSonics, 2024-02-06:
@sys-met  I am facing the same issue. you mentioned that "The problem was that I forgot the cmd_vel setting. I set it up with twist_mux and it works fine." could you please clarify more in details.

hoangnhuet, 2024-08-06:
hi, thanks for your great work. Iam a new in ros2 and i am facing the same problem. Could you tell me more about how can i fix this cmd_vel (the params file location, ...) setting. iam really appreciate.