ros-navigation/navigation2 #4033: Feature request: Graceful cancellation of `follow_path` actions

Rayman, 2024-01-05:
## Feature request
Ability to let controllers handle cancelations gracefully

#### Feature description
At the moment when the `follow_path` action is cancelled, a zero velocity is immediately published ([see the code](https://github.com/ros-planning/navigation2/blob/305f7e039f1183b94981ec6a7f294f0ed7b89d07/nav2_controller/src/controller_server.cpp#L463) and the figure below).
We have a big heavy robot that needs a lot of time to slow down so we want to keep following the current path while decelerating within the deceleration limits.

With move base flex, we could override the `cancel` method and return when the cancellation was performed (see 
https://github.com/magazino/move_base_flex/issues/171)
Is there a way to achieve this graceful cancel or can we add the same API to Navigation2?

[image]

Timple, 2024-01-05:
The idea is to decelerate by dynamic limits _whilst tracking the path_. 

When following a turn, it would be undesired to slow down in a straight line upon a cancel action, which I think/guess is what the `nav2_velocity_smoother` does?

SteveMacenski, 2024-01-05:
I also don't think that's actually a generically safe solution? Deceleration in turns could cause the kinds of "throwing loads off my robot top" that the instantaneous braking could due to centrifugal forces. I imagine as well there are techniques where reducing the velocity scale will actually make it deviate off the path due to its internal modeling (thinking MPC/MPPI which are maintaining some state) 

This isn't my opposition to your request - just throwing out some additional constraints perhaps that weren't considered giving me pause

SteveMacenski, 2024-01-05:
> When following a turn, it would be undesired to slow down in a straight line upon a cancel action, which I think/guess is what the nav2_velocity_smoother does?

The Vel Smoother will ramp down from the last request to the zero velocity. It won't necessarily be a straight line. If there's some non-zero angular component, that will be ramped down too by its own deceleration constraints

Rayman, 2024-01-08:
When the robot is paused during operation (no emergency stop)
```
target_x_acc: 0.25 m/s^2
target_x_vel: 3.8333 m/2. 
```
That means it takes ~15 seconds to fully slow down. The braking distance is (if I remember correctly)
```
s = v^2/2/a = 29.4 m
```
Driving 30 meters without following the path will not work for us. Image being on a straight line just before a turn, the robot would continue driving straight and drive 30 meters off the path.

I think our requirements are a bit different than a lot of other users, but it would be nice if we could stick to a common standard.

Does that command velocity smoother have access to the path? That would solve the problem. Otherwise the controller could signal that a cancel was handled so we can have ~15 seconds of extra control time.

SteveMacenski, 2024-01-08:
Ah ok. I understand the need then better. The velocity smoother doesn't have access to the path, but I suppose it could be made to have it. The main question I'd have there is how to follow the path in that cancel ramp down? 

I think a path forward here would be for you to propose a design to enable this for us to discuss. Once we agree, I'm also agreeing to merge that in once you put up a PR. 

A gut instinct would be to have a new Controller plugin API for handling cancellations. I'm not 100% sure what that would entail internally off hand, but this is clearly something you've thought about so I'll leave that to your experience :-). The Controller Server can call that cancel API when a cancel request is formed. Or, I suppose it could flip a state and call the cancel API instead of the compute API in the `while` loop  so we still perform all the other server's work without duplication. 

The only request I'd have beyond what you'd naturally do anyway would be to implement such a cancellation ramp down for +1 of the existing controllers so that there is an example of it to build from with the others. If you're already going to modify one of them for your needs that you use currently, I'd ask that you do a second just to help what otherwise I'll have to come back and do (or if you want to be very generous, do them all, which I would appreciate :-) )

SteveMacenski, 2024-01-30:
@Rayman any update? :-)

Rayman, 2024-02-01:
I'm trying to figure out which API we could use. I have nav2 building from souce and I'm trying to modify the pure pursuit planner to handle cancellation. But I'm bumping against some issues with the simulator (and another project which wants my attention)

At the moment I have
```cpp
// Signal the controller that a cancellation is requested. By default returns true to indicate an immediate cancellation is OK.
bool cancel()
```
If you return false you get an extended gracefull cancel. `computeVelocityCommand` should raise an `CancellationCompleted` exception. When the cancellation has completed successfully.

But I'm not really happy with raising an expection to indicate a successful cancellation, so I'm thinking some other methods.

SteveMacenski, 2024-02-01:
Maybe setting something in the is_cancel_requested conditional? I think there are 2 places that jump out to me:

- The `is_cancel_requested` conditional, add a new cancellation section before we `terminate_all()` like `computeAndPublishCancelVelocity`.
- Or setting some state so that we continue in the main look, but set a flag instead of compute control, to compute _cancel_ controls (which by default we can set in the controllers to just send `0`.  That has the benefit of keeping the current control rate object, updating global paths, getting robot pose to send, and such all the same.  If we rearrange the main `while` loop to have the costmap current, update global path checks before this condition, we could actually keep the same loop if we `continue` or set an `if/else` state for cancel vs control as to which branch to execute before the shared sleep. 

The second seems like the most reasonable to share as much as possible and keep the cancel control have the same loop rate and behavior as main control and have just some `computeVelocityCommands` for cancelation: `computeCancelCommands`. That API should return a state if its "done" so we can use that to break out of the cancel state and finally `terminate` and return

closed 2024-03-28 by SteveMacenski