ros-navigation/navigation2 #4042: Include option to use PointCloud Transport in Costmap2D

SteveMacenski, 2024-01-09:
Per https://github.com/ros-perception/point_cloud_transport, we can now use compressed pointcloud streams in ROS 2. We can enable those in the Costmap layers that ingest pointcloud sensor streams (i.e. obstacle, voxel) and the collision monitor and show that this reduces network traffic without composed nodes to keep up better (and CPU impacts).

We would want sensor drivers and depth pipelines to have the option to output this as well to be able to ingest in Nav2 out of the box. Rviz2 + Gazebo would be ideal as well to publish PC2 from Gazebo into Rviz without as much latency 

CC @ahcorde

Combuster54, 2024-09-24:
Hi, could you provide more info? I would like to contribute if it´s possible.

xFranv8, 2024-09-25:
I'm taking this issue. Thanks!

SteveMacenski, 2024-12-10:
@xFranv8 any updates?

elsayedelsheikh, 2025-01-19:
I think we have a similar case to `image_transport` package, where `image_transport` and `point_cloud_transport` don't have LifecycleNode support

- https://github.com/ros-perception/image_common/issues/108
- https://github.com/ros-perception/image_common/pull/304

https://github.com/ros-perception/image_common/blob/b021202a42d7bc3ff3215237ded251a59c282da5/image_transport/include/image_transport/publisher.hpp#L72-L77  

https://github.com/ros-perception/point_cloud_transport/blob/0350a390bdd8cc39c509b71614a4c613457ce909/point_cloud_transport/include/point_cloud_transport/publisher.hpp#L59-L64

Should we wait till it gets implemented in `point_cloud_transport`? or/ work around it?

SteveMacenski, 2025-01-21:
Do they need lifecycle support if we create them in the configure / activate methods of our lifecycle servers? Note in Costmap2D we use Message Filters, which has an unsubscsribe/subscribe option -- I'm not sure if/how you can or should use both message filters and image / PC2 transport, but I imagine you can by daisy chaining callbacks. If so, then that unsubscribe/subscribe feature would give you virtual lifecycle support for our needs here, even if not supported by those libraries.

Optionally: you could choose to implement it in the transport pipelines as part of this ticket, I'm sure the maintainers would love the help and further have impact in the community! But, I don't think it is largely necessary, since the transports are used within costmap layers or within operations that themselves are happening in lifecycle transitions. We don't try to make the subscribers until we're in the configure state (though between configure and activate... the [message filters subscribe](https://github.com/ros-navigation/navigation2/blob/main/nav2_costmap_2d/plugins/obstacle_layer.cpp#L726) resolves that for us for now -- if we can daisy chain and use both; don't return callback until a TF transform exists **and** use the compression pipelines)

elsayedelsheikh, 2025-01-23:
Perfect!
Moving on with [PCL Transport SubscriberFilter variant](https://github.com/ros-perception/point_cloud_transport/blob/0350a390bdd8cc39c509b71614a4c613457ce909/point_cloud_transport/include/point_cloud_transport/subscriber_filter.hpp#L63) for the **Obstacle layer**

Tasks:
- [ ] obstacle_layer
- [ ] voxel_layer
- [ ] collision_monitor
- [ ] Add transport method configuration with "raw" as default
- [ ] ... missing something ? 

Will update [nav2_minimal_turtlebot_simulator](https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/tree/main) to publish PCL for testing

Thanks :)

SteveMacenski, 2025-01-23:
Conveniently, the voxel layer uses the obstacle layer's base for subscriptions, so its a 2 for 1!

I think also [STVL](https://github.com/SteveMacenski/spatio_temporal_voxel_layer)/[NPVL](https://github.com/SteveMacenski/nonpersistent_voxel_layer) would be good as well, while you're at it!  

>  Add transport method parameter with "raw" as default ?

I think that would be good

Also add into your tasks for documentation updates to the layer's configuration file, migration guide, and best yet if you wanted to write a quick tutorial about how to use it (and some metrics about how it improves performance - possibly?)

DasariChaitanya, 2025-01-23:
Hi everyone,

I've been working on this topic for a few days now and didn't realize others were also on it. I should have posted earlier. I have not noticed there's a Subscribe Filter variant in PCL transport that might be a better implementation than what I'm currently using.

@elsayedelsheikh, let me know if you'd like any assistance.

elsayedelsheikh, 2025-01-27:
Hi @DasariChaitanya 
If you're still up for this, I could use some help 😁
Check the updated TODO list

DasariChaitanya, 2025-01-27:
Hi @elsayedelsheikh 
Yes, let's get started on the tasks. 

1. Did we open an issue for the lifecycle node support for `point_cloud_transport`? 
2. Another potential issue I came across was the [subscriber vector ](https://github.com/ros-navigation/navigation2/blob/6e1c85eba70ed9800b69886c15d73268d60f46e9/nav2_costmap_2d/include/nav2_costmap_2d/obstacle_layer.hpp#L235) class is not compatible with class of [PCL Transport SubscriberFilter variant](https://github.com/ros-perception/point_cloud_transport/blob/0350a390bdd8cc39c509b71614a4c613457ce909/point_cloud_transport/include/point_cloud_transport/subscriber_filter.hpp#L63)
3. I have not looked at publishers yet. But I think it would be better to use all available transports for outputs

SteveMacenski, 2025-03-14:
Hi! Do you have any updates on this work? I know some great progress was made and I wanted to know when we could expect a PR to add it into the stack!

SteveMacenski, 2025-03-17:
@elsayedelsheikh you may want to ping the maintainer again to answer your question!