ros-navigation/navigation2 #4061: MPPI - UpBoard navigation error

rafal-gorecki, 2024-01-19:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- ROS2 Version: Humble
- Version or commit hash: ros-humble-nav2-mppi-controller	1.1.12-1jammy.20231205.201627
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
Difficult due to requirement of [UpBoard specs](https://up-board.org/up/specifications/). Generally all my work is encapsulated inside [Docker container](https://github.com/husarion/rosbot-autonomy/tree/foxglove-update).

#### Behavior
In my case i have two robots with diffrent procesor architecture: one is RPi4 and second UpBoard. For both Robots running this [Docker container](https://github.com/husarion/rosbot-autonomy/tree/foxglove-update) on both robots. The first robot with RPi4  working fine, the second robot with UpBoard throws error which doesn't tell much:

```bash
[ERROR] [component_container_isolated-1]: process has died [pid 107, exit code -4, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container -r __ns:=/rosbot2pro --params-file /tmp/launch_params_osr30n1p --params-file /tmp/launch_params_nu2yugfv -r /tf:=tf -r /tf_static:=tf_static']
```

Running rpp controller works fine. Is it possible that the constructed instructions are not executed correctly on another processor? I'm asking for suggestions and I'm ready to test something.

SteveMacenski, 2024-01-19:
That’s possible. Try removing the avx and other compiler optimization flags from MPPI. It will likely run slower. Also if you can reproduce with source builds, using Nav2’s traceback tutorial on navigation.ros.org would be helpful info to see where its happening

SteveMacenski, 2024-01-30:
Closing - 2 weeks and multiple reminders without a response.

rafal-gorecki, 2024-02-05:
Sorry for delay. As per your request, I did a traceback and got this error from the controller.

```bash
Thread 1 "controller_serv" recived signal SIGILL, Illedgak ubstruction. 0x00007f3df0116005 in class_loader::impl::MetaObject<nav2_mppi_controller::MPPIController, nav2_core2::Controller>::create() const () from /ros2_ws/install/nav2_mppi_controller/lib/libmppi_controller.so
```

 I'm currently building mppi, with different flags, but the controller doesn't build every time, so I haven't been able to check which compilation flag causes the error.

rafal-gorecki, 2024-02-06:
Thank you Steve for quick response.

I'm having problems with backtrace. 
1. I remove all other `add_compile_options` and add `add_compile_options(-g)` .
2. Add `prefix=['xterm -e gdb -ex run --args'],` to controller server in `naviagation.launch.py`.
3. I also add `backward_ros` to mppi package.
4. Rebuild  (`colcon build --symlink-install`) and run project and i still see the same error in xterm.

If you know why this is happening, I would appreciate your advice.

SteveMacenski, 2024-02-06:
You must also compile MPPI with debug flags. You shouldn't need to add `backward_ros` nor mess with the compilation flags otherwise. Just add that line, don't remove anything else.

rafal-gorecki, 2024-02-07:
Hello again Steve,
I managed to run backtrac. There is not much more information as I received the following message.

```bash
Thread 1 "controller_serv" recived signal SIGILL, Illedgak ubstruction.
class_loader::impl::MetaObject<nav2_mppi_controller::MPPIController, nav2_core2::Controller>::create (this=<optimized out>)
at /opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218
218 return new C;
```

Can you help me with this and is there anything I can do to help?

rafal-gorecki, 2024-02-08:
Hello Steve,
Thank you for your suggestions and I received them accordingly.

Backtrace after removing all `add_compile_options` (without `-g) shows:
```bash
Thread 1 "controller_serv" recived signal SIGILL, Illedgal instruction.
class_loader::impl::MetaObject<nav2_mppi_controller::MPPIController (this=0x555555c161a0)
at /ros2_ws/src/nav2_mppi_controller/include/nav2_mppi_controller/controller.hpp:46
46 MPPIController() = default;
```

CPU flags:
```bash
flags: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb pti ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid tsc_adjust smep erms dtherm ida arat md_clear
```

SteveMacenski, 2024-02-08:
At this point, you'll need to do some more debugging on your own, I don't have your hardware and these errors don't give me alot to go off of. I'd recommend just moving to more mainstream hardware. You'll likely spend more of your time trying to figure this out than the board is worth if its for a hobbyist project - and ultimately that CPU might be too weak to even effectively run MPPI.

rafal-gorecki, 2024-02-08:
Thank you for all your replies.  In my free time I will try to do something else.  I'll let you know if I discover something.  I'm closing the issue for now.

closed 2024-02-08 by rafal-gorecki

SteveMacenski, 2024-02-08:
Sorry I couldn't give you the exact blueprint :(