ros-navigation/navigation2 #4066: Homologate Smac & MPPI: getInflationLayer (easy)

SteveMacenski, 2024-01-25:
Recently a new `static getInflationLayer()` method was added to the inflation layer to get the inflation layer out of a stack of layers in the layered costmap. This was added by a PR to add additional intelligence to the Smac Planner in heuristic computation for non-circular footprints. This logic was largely lifted from MPPI's obstacle critic to find the inflation layer, so rather than putting it in Smac, it was added to a third party location so they can be shared (i.e. homologated).

The task is to take this new function and apply it to the Obstacle critic: https://github.com/ros-planning/navigation2/blob/main/nav2_mppi_controller/src/critics/obstacles_critic.cpp#L55-L101

jncfa, 2024-01-26:
> Recently a new `static getInflationLayer()` method was added to the inflation layer to get the inflation layer out of a stack of layers in the layered costmap. This was added by a PR to add additional intelligence to the Smac Planner in heuristic computation for non-circular footprints. This logic was largely lifted from MPPI's obstacle critic to find the inflation layer, so rather than putting it in Smac, it was added to a third party location so they can be shared (i.e. homologated).
> 
> The task is to take this new function and apply it to the Obstacle critic: https://github.com/ros-planning/navigation2/blob/main/nav2_mppi_controller/src/critics/obstacles_critic.cpp#L55-L101
> 
> New Fn introduced here: #4067

I assume you would also want to replace it here, correct? https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/include/nav2_smac_planner/utils.hpp#L82-L97

I can make a quick MR for this if you want

SteveMacenski, 2024-01-26:
Sure!

closed 2024-01-26 by SteveMacenski