ros-navigation/navigation2 #4081: [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...

GPrathap, 2024-01-29:
- Operating System:
     Ubuntu 22.04 LTS and Humble 

- DDS implementation:
     FastDDS

- Client library (if applicable):
     rclcpp

Suddenly, I am getting this error. 

To replicate: 
 I have used two different ways:

      ros2 action send_goal --feedback /navigate_to_pose nav2_msgs/action/NavigateToPose '{pose: {header: {frame_id: "map"}, pose: 
     {position: {x: 20.0, y: -5.0, z: 0.0}, orientation: {x: 0.0, y: 0.0, z: 0.0, w: 1.0}}}}'

And try to pass [navigate_to_pose_w_replanning_and_recovery](https://github.com/ros-planning/navigation2/blob/humble/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml) with pose information. Also, I've tried [navigate_through_poses_w_replanning_and_recovery](https://github.com/ros-planning/navigation2/blob/humble/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml) even for that I am getting the same error. 

However, in both cases, I am getting this error 

[component_container_isolated-1] [INFO] [1706550801.492917075] [bt_navigator]: Begin navigating from current location (10.74, -9.79) to (20.00, -5.00)
**[component_container_isolated-1] [ERROR] [1706550801.493022499] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...**
[component_container_isolated-1] [WARN] [1706550801.609195257] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 8.6172 Hz
[planner_server-10] [WARN] [1706550801.633046650] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 7.1490 Hz
**[component_container_isolated-1] [ERROR] [1706550801.893112726] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...**
[component_container_isolated-1] [INFO] [1706550801.994502174] [controller_server]: Received a goal, begin computing control effort.
**[component_container_isolated-1] [ERROR] [1706550803.093334453] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...**
[component_container_isolated-1] [WARN] [1706550803.297329429] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 4.9030 Hz
[planner_server-10] [WARN] [1706550803.297992218] [planner_server]: Planner loop missed its desired rate of 20.0000 Hz. Current loop rate is 4.8908 Hz
[component_container_isolated-1] [ERROR] [1706550803.593087705] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[component_container_isolated-1] [INFO] [1706550803.695279540] [controller_server]: Passing new path to controller.
[component_container_isolated-1] [ERROR] [1706550804.793296847] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[component_container_isolated-1] [ERROR] [1706550804.993069433] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...

SteveMacenski, 2024-01-29:
I tested the binaries on Humble and I see no issues on my end through rviz or commanding through the action CLI as you did. I'd recommend looking into your DDS configurations. Have you tried Cyclone?

GPrathap, 2024-01-30:
Hi Steve, 

Yes I tried both, by setting 

                  export RMW_IMPLEMENTATION=rmw_fastrtps_cpp

and 

                   export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

GPrathap, 2024-01-30:
Is there a way to debug this issue? I am kind of lost (

GPrathap, 2024-01-30:
I've resolved the issue, 

This unexpected issue is due to mistakenly adding `nav2_planner` and a normal node and part of a composable node. After I removed the normal node everything worked fine.  

I am not sure if this is a bug or not. However, if the same node lunch two different ways, the system somehow should know about it.

closed 2024-01-30 by SteveMacenski