ros-navigation/navigation2 #4086: Nav2 modules use 100% of CPU

jesusramondovale, 2024-01-31:
## Bug report

**Required Info:**
- Machine: 
   - Single Board Computer -> Debix Model A (2GB RAM) 
- Operating System:
  - Ubuntu 22.04 
- ROS2 Version:
  - Humble
- DDS implementation:
  - Fast-RTPS

#### Expected behavior
I'm using a SBC Debix Model A with an quad-core ARM Cortex-A53 processor : NXP i.MX 8M Plus (2.3TOPS NPU) (aarch64) and using htop I can see that all 4 cores reach a 100% usage only by launching the nav2 modules with 

```
IncludeLaunchDescription(
            PythonLaunchDescriptionSource([os.path.join(
            get_package_share_directory('nav2_bringup'), 'launch'),'/bringup_launch.py']),
            launch_arguments={'map': map_file_path,
                              'autostart': 'True'}.items()
                
        ),
```

[image]

#### Actual behavior
Navigation works but when I increase the maximum speed values (it's a basic 2-wheeled differential robot) it starts going very bad

SteveMacenski, 2024-01-31:
Nav2 is not a single threaded application. I'd have to review the stack, but I think its ~10-20 threads, depending on what you have running. 120% of a core seems about right and inline with [our published metrics](https://arxiv.org/pdf/2305.09933.pdf). Frankly, its **amazing** we only use 1.2 of those weak, low power ARM cores to run all of autonomous navigation!

That is not "only" running Nav2, I see some accounts daemon, microros, rosbridge themselves summing up to 1.6 cores (more than Nav2!). 

This is not a safe configuration. You're thrashing the CPU and memory, I would not attempt to operate a robot under these compute conditions unless it was a small toy robot that if it had problems in an R&D lab would be no big deal. You need more compute.

closed 2024-01-31 by SteveMacenski

SteveMacenski, 2024-02-01:
I don't understand what you mean or what behavior you're trying to remove.

jesusramondovale, 2024-02-02:
I'd like to know how would you try to reduce at maximum the CPU usage (talking only about Nav2 modules)