ros-navigation/navigation2 #4088: [for help] a frequent NullPtr-accessed bug happening during navigation use

GoesM, 2024-02-01:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu22.04
- ROS2 Version:
  - humble
- Version or commit hash:
  -  the latest
- DDS implementation:
  -  defaulted

#### Steps to reproduce issue

I always find an asan-report about a NullPtr-accessed bug during my daily use.

Following is how I launched navigation2.
```sh
#!/bin/bash
export ASAN_OPTIONS=halt_on_error=0:new_delete_type_mismatch=0:detect_leaks=0:log_pah=asan
source install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=True use_rviz:=False use_composition:=False 
```

#### Expected behavior

#### Actual behavior

Asan-report like following: 
```asan
=================================================================
==3031718==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7f5a74fea816 bp 0x61700000e080 sp 0x7f5a72dfa6b0 T4)
==3031718==The signal is caused by a READ memory access.
==3031718==Hint: address points to the zero page.
    #0 0x7f5a74fea816 in eprosima::fastrtps::rtps::EDP::unpairWriterProxy(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::GUID_t const&, bool) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x3ea816) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #1 0x7f5a74feb425 in eprosima::fastrtps::rtps::PDP::remove_remote_participant(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::ParticipantDiscoveryInfo::DISCOVERY_STATUS) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x3eb425) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #2 0x7f5a74fecbad in eprosima::fastrtps::rtps::PDPListener::onNewCacheChangeAdded(eprosima::fastrtps::rtps::RTPSReader*, eprosima::fastrtps::rtps::CacheChange_t const*) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x3ecbad) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #3 0x7f5a74e7b8df in eprosima::fastrtps::rtps::StatelessReader::change_received(eprosima::fastrtps::rtps::CacheChange_t*) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x27b8df) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #4 0x7f5a74e7bcf1 in eprosima::fastrtps::rtps::StatelessReader::processDataMsg(eprosima::fastrtps::rtps::CacheChange_t*) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x27bcf1) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #5 0x7f5a74e7ff1f in eprosima::fastrtps::rtps::MessageReceiver::process_data_message_without_security(eprosima::fastrtps::rtps::EntityId_t const&, eprosima::fastrtps::rtps::CacheChange_t&) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x27ff1f) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #6 0x7f5a74e89a7a in eprosima::fastrtps::rtps::MessageReceiver::proc_Submsg_Data(eprosima::fastrtps::rtps::CDRMessage_t*, eprosima::fastrtps::rtps::SubmessageHeader_t*) const (/opt/ros/humble/lib/libfastrtps.so.2.6+0x289a7a) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #7 0x7f5a74e8b6c7 in eprosima::fastrtps::rtps::MessageReceiver::processCDRMsg(eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::CDRMessage_t*) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x28b6c7) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #8 0x7f5a74e9134e in eprosima::fastrtps::rtps::ReceiverResource::OnDataReceived(unsigned char const*, unsigned int, eprosima::fastrtps::rtps::Locator_t const&, eprosima::fastrtps::rtps::Locator_t const&) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x29134e) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #9 0x7f5a74f18495 in eprosima::fastdds::rtps::UDPChannelResource::perform_listen_operation(eprosima::fastrtps::rtps::Locator_t) (/opt/ros/humble/lib/libfastrtps.so.2.6+0x318495) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #10 0x7f5a74f130ba  (/opt/ros/humble/lib/libfastrtps.so.2.6+0x3130ba) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10)
    #11 0x7f5a776dc252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252) (BuildId: e37fe1a879783838de78cbc8c80621fa685d58a2)
    #12 0x7f5a77294ac2 in start_thread nptl/./nptl/pthread_create.c:442:8
    #13 0x7f5a77326a3f  misc/../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/opt/ros/humble/lib/libfastrtps.so.2.6+0x3ea816) (BuildId: 70b21fef9289ed9b8f7ede94d4ece97ca2f1cc10) in eprosima::fastrtps::rtps::EDP::unpairWriterProxy(eprosima::fastrtps::rtps::GUID_t const&, eprosima::fastrtps::rtps::GUID_t const&, bool)
Thread T4 created by T0 here:
    #0 0x5556f0e567dc in __interceptor_pthread_create (/home/****/ROS_ws/nav2/install/nav2_smoother/lib/nav2_smoother/smoother_server+0x937dc) (BuildId: 4b98fb0821be2c781667376adf7ef527adf20f57)
    #1 0x7f5a776dc328 in std::thread::_M_start_thread(std::unique_ptr<std::thread::_State, std::default_delete<std::thread::_State> >, void (*)()) (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc328) (BuildId: e37fe1a879783838de78cbc8c80621fa685d58a2)

==3031718==ABORTING
```

#### Additional information

----

During my daily use, such bug not only happened to `nav2_smoother`, but also happened to `nav2_bt_navigator`, `nav2_velocity_smoother` and many other packages.

Since using Navigation2, I have seen such Asan-report frequently. But I have not been very sure whether this bug is caused by navigation2 or other ros2-repositories. So I'd appreciate if anyone could help any for this issue.

SteveMacenski, 2024-02-01:
It looks like Fast-DDS, you should file something over there. Have you tried using Cyclone and seen the same issue?

SteveMacenski, 2024-02-18:
Please follow up if there’s a way we can help! I think though taking this up with the DDS maker or RMW repo would be best. If there’s something we need to adjust in Nav2, let us know!

id definitely recommend at least just trying another DDS just as a sanity check that this is unique to one or something more broadly an issue in rclcpp worth escalating to those maintainers. If you need some help with that, let me know, but closing since the trace isn’t something in Nav2 and doesn’t seem necessarily actionable for this issue tracker

closed 2024-02-18 by SteveMacenski