ros-navigation/navigation2 #4104: Smac Planner Hybrid Issue

anath93, 2024-02-06:
- Operating System: Ubuntu 20.04
- ROS2 Version: Galactic
- Version or commit hash: I saw an issue previously and built the Smac planner from main to galactic.

#### Steps to reproduce issue
[image]

[image]

[image]

[image]

#### Expected behavior

The planner should re-route by seeing the obstacle when raytrace is in range. 

#### Actual behavior

It keeps computing ending up burning in the bin, its diff drive robot. But it does stop and says collision detected and goes in recovery mode.

SteveMacenski, 2024-02-06:
Its unclear from your diagram: is that present in the global costmap used for planning or the local costmap used for control, or both? 

That's doesn't look like a lethal obstacle to me (yeah?), that's just a higher cost region as evidenced that there is no inflation around it. The cost functions weight distance with cost, so in this situation, it was better to go through 1-2 cells of higher cost than to diverge dozens.

anath93, 2024-02-06:
Yes it is present in the global costmap, and its not lethal, as you said higher cost but shouldn't Smac Hybrid planner plan around it  based on the cost penalty with conjunction of global costmap ?
This is my below global and local costmap setup so far, 
[image]
[image]
[image]

SteveMacenski, 2024-02-06:
> Yes it is present in the global costmap, and its not lethal, as you said higher cost but shouldn't Smac Hybrid planner plan around it based on the cost penalty with conjunction of global costmap ?

Not necessarily, there are trade offs beyond just cost, like shortest length. We couldn't want a robot necessarily to add 100m to a path to stay in `0` cost space over entering 1 cell that is `1` cost, right? So we need to decide on a trade off between minimum length and reasonable cost to travel. We do that with the traversal function of `cost_to_travel = cell_length * (1.0 + cost / 254 * weight)`. That makes us balance the cost to travel by both the length of the primitive with the normalized, weighted cost. A higher cost weight would make traveling in higher cost spaces more expensive. [Figure 5 of this paper](https://arxiv.org/pdf/2401.13078.pdf) highlights how the change in cost penalty changes behavior.

Looking at your example, it seems like the extra ~+2m to go around it isn't proportionate to the extra cost from such a small obstacle (its only 1-2 pixels thick). I might expect a different result if your virtual obstacle there was filled. What you show is more or less expected and good behavior. If you wanted that are to be not traversed at all, you should mark it with lethal cost via whatever custom mechanism you've added that with

Note: In general, please don't post pictures of parameters, include them as code snippets for the sections that are important. But also note that nothing you showed is really relevant.

closed 2024-02-06 by SteveMacenski