ros-navigation/navigation2 #4107: Static layer plugin ignores the map_topic parameter and reverts to default behaviour

sagarvl96, 2024-02-07:
## Bug report

**Required Info:**

- Operating System:
     Ubuntu 22.04.3 LTS
- ROS2 Version:
     ROS2 Humble base docker
- Version or commit hash:
      ros-humble-navigation2  1.1.12-1jammy.20240126.081240
- DDS implementation:
       rmw_cyclonedds

#### Steps to reproduce issue
   ```
static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: True
        map_topic: /refined_map
```
   Set the map_topic parameter of Static Layer to a custom map topic (/refined_map).

#### Expected behavior
    Subscribing to the map topic (/refined_map) with transient local durability

#### Actual behavior
     Subscribing to the map topic (/map) with transient local durability
     (Still subscribes to the global map topic (/map) and ignores the topic mentioned as a parameter)

SteveMacenski, 2024-02-07:
Are you sure that you rebuilt your workspace after making the parameter change so the new parameters are being reflected in your launch file? 

I'm looking at the code, and we don't seem to have a default name for the map topic (https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/plugins/static_layer.cpp#L134-L150), so if its even getting `map` its being set from somewhere in your configuration file(s) or you're not using the files you think you are. The other possible error is if you have `map_topic` defined in a parent namespace before the plugin to `map`, that'll be possibly pulled in here if the map topic isn't picked up here for whatever reason.

I'm curious if you set the QOS to `false` instead of `true` if you see that change happening in the log. If not, then I think your problem is that param file isn't being used or wasn't rebuilt.

sagarvl96, 2024-02-08:
I thank you for your quick response. I checked the parameters and I saw that there was already a map_topic parameter. I was stupid to not check this before. I will close this issue as it has been resolved.

sagarvl96, 2024-02-13:
Hello Steve,
My custom map topic has a different orientation of its origin. But the [processMap function in the static layer plugin](https://github.com/ros-planning/navigation2/blob/ab9b1010d7a126c0c407cd09228d97068646ee4c/nav2_costmap_2d/plugins/static_layer.cpp#L176) only changes the origin's position. How do I change the origin's orientation wrt to my custom map topic? And what is the default orientation and position of the origin?? Your response in this matter would be highly appreciated.
Thank you!

SteveMacenski, 2024-02-13:
Ask on Robotics Stack Exchange. Don't reopen old tickets on other topics.

closed 2024-02-13 by SteveMacenski