ros-navigation/navigation2 #4113: rename possible_inscribed_cost into possible_circumscribed_cost in collision_checker.setFootprint

alanxuefei, 2024-02-13:
## Bug report

The name of `possible_inscribed_cost ` in [`GridCollisionChecker::setFootprint`](https://github.com/ros-planning/navigation2/blob/ab9b1010d7a126c0c407cd09228d97068646ee4c/nav2_smac_planner/src/collision_checker.cpp#L54) may not be accurate. 

```
void GridCollisionChecker::setFootprint(
  const nav2_costmap_2d::Footprint & footprint,
  const bool & radius,
  const double & possible_inscribed_cost)
```
Because [nav2_smac_planner](https://github.com/ros-planning/navigation2/blob/ab9b1010d7a126c0c407cd09228d97068646ee4c/nav2_smac_planner/src/smac_planner_hybrid.cpp#L220C1-L223C42) and [smac_planner_lattice](https://github.com/ros-planning/navigation2/blob/ab9b1010d7a126c0c407cd09228d97068646ee4c/nav2_smac_planner/src/smac_planner_lattice.cpp#L185) use circumscribed radius cost (outside circle) instead of inscribed radius cost (inside circle).

```
  _collision_checker.setFootprint(
    _costmap_ros->getRobotFootprint(),
    _costmap_ros->getUseRadius(),
    findCircumscribedCost(_costmap_ros));
```

Inscribed radius (inside circle) may not be sufficient for conduct collision check.

SteveMacenski, 2024-02-13:
I think this is correct (at least in my view), but I agree perhaps it could be renamed more clearly. `possible_collision_cost`?

We use the circumscribed radius as a check to see if its possible that any point on the robot might be in collision. I agree `inscribed` is an odd choice when I wrote that. We should update MPPI's Cost and Obstacle critics' variables of similar use while we're at it!

alanxuefei, 2024-02-13:
I will create a PR within this week.

closed 2024-02-17 by SteveMacenski