ros-navigation/navigation2 #4121: simple action server bug

iceuw, 2024-02-19:
## Bug report

I guess there is some wrong with this line of code in https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L577

The argument `std::shared_ptr<rclcpp_action::ServerGoalHandle<ActionT>> handle` should change to `std::shared_ptr<rclcpp_action::ServerGoalHandle<ActionT>>& handle`. Otherwise, the function `terminate` won't free the 'handle' correctly. 

And if you send a wrong goal to a running action server, the code will terminate pending goal with not free the pending goal. And then the function `is_cancel_requested` in  https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp#L449 will always return false cause the pending handle is not a nullptr.

iceuw, 2024-02-22:
>how you send a "wrong goal" and how it terminates... itself
I found this problem when I sent a goal to "navigate_through_poses", But it was terminated by https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/src/navigators/navigate_through_poses.cpp#L195. Then I tried to cancel current goal and send a new goal. But I failed. The server has still been running on the goal I want to cancel.

mikeferguson, 2024-02-22:
@iceuw is correct here - reset() is only going to affect that instance, and decrement the reference count (if the count were to go to zero, then yes, the actual object would be destructed - but in this case, there is still a shared_ptr outside the function call).

If reset() cleared out ALL instances of the shared_ptr - then shared_ptrs would actually be pretty useless (because anywhere in the code, I could destruct the instance to which you are still holding a reference to, and you would have to continuously check if your shared_ptr was now nullptr)

Some more examples on stack overflow: https://stackoverflow.com/questions/21589595/does-using-reset-on-a-stdshared-ptr-delete-all-instances

closed 2024-02-23 by SteveMacenski