ros-navigation/navigation2 #4132: Controller Server stuck after Aborting Handle

mrmara, 2024-02-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.2 LTS x86_64
- ROS2 Version:
  - Humble debian
- Version or commit hash:
  - commit 54a45cb7eed58aaa27921ce1af0d54c480de3184
- DDS implementation:
  - cyclone

#### Steps to reproduce issue

Use only keepout_filter and inflation_layer as local_costmap plugins
Send a navigate_to_pose goal to the behavior tree. 
Make the robot not move, for example due to a collision_monitor stop polygon.
Wait until progress checker hits.
Controller server will not start again.

##### Another way to reproduce the issue.

Use obsacle_layer, keepout_filter, inflation_layer as local_costmap plugins
Send a navigate_to_pose goal to the behavior tree.
Disable obstacle layer with dynamic reconfigure to obstacle_layer.enable param
Make the robot not move, for example due to a collision_monitor stop polygon.
Wait until progress checker hits.
Controller server will not start again until obstacle_layer is enabled again.

#### Expected behavior
Controller should retry navigation for the prescribed max attempts number
#### Actual behavior
Controller server remains stuck in aborting previous action.
FollowPath keeps logging "Aborting handle" (simple_action_server.hpp - method terminate -line 588)
According to debug logs inside simple_action_server the new goal is moved to a pending slot because the old goal is still active but it cannot terminate Somehow the obstacle layer unblock this situation.

SteveMacenski, 2024-02-21:
Please provide logs.

> Use only keepout_filter and inflation_layer as local_costmap plugins

This is an immediate thing I can flag, why have an inflation layer if there's nothing being inflated? If there's nothing in the layers, then nothing is being done in the costmaps. The costmap filters are in the separate filters API. So this sounds like a misconfiguration somehow getting us into a weird state. 

Throw a log here: https://github.com/ros-planning/navigation2/blob/main/nav2_controller/src/controller_server.cpp#L468-L471 is that where its holding up? If not, please poke around and try to get more information. This isn't alot to go off of. Where is that abort being called from, for instance? That might help point back to where this is happening.

mrmara, 2024-02-22:
Thank you for your response.

Let me clarify something, my normal configuration has the obstacle layer and so the inflation layer inflates the obstacles. Sometimes I disable it with dynamic reconfigure and navigate just with collision monitor. When opening the issue I described both methods to reproduce it just to isolate the problem and prove that the stuck condition of the controller_server is linked to the obstacle layer.
The controller server remains stuck exactly where you said but I do not see any strange log on local costmap.

<details>
<summary> LOGS </summary>
[controller_server-1] [WARN] [icn04.controller_server]: [follow_path] [ActionServer] Aborting handle
. (warn_msg() at /home/mrmara/workspace/install/nav2_util/include/nav2_util/simple_action_ser
ver.hpp:630)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be valid (computeCon
trol() at /home/mrmara/workspace/src/navigation2/nav2_controller/src/controller_server.cpp:40
8)
[... 200 lines cut ...]
<details>
<summary> LOGS </summary>
```
[DEBUG] [icn04.controller_server]: Waiting for costmap to be
 valid (computeControl() at /home/mrmara/workspace/src/navigation2/nav2_con
troller/src/controller_server.cpp:408)
[controller_server-1] [INFO] [nav2_costmap_2d]: Checking plugin obstacle_layer, is
current 0 || isNOTenabled 1 == 1 (isCurrent() at /home/mrmara/workspace/src
/navigation2/nav2_costmap_2d/src/layered_costmap.cpp:266)
[controller_server-1] [INFO] [nav2_costmap_2d]: Checking plugin inflation_keepout,
 iscurrent 0 || isNOTenabled 0 == 0 (isCurrent() at /home/mrmara/workspace/
src/navigation2/nav2_costmap_2d/src/layered_costmap.cpp:266)
[... cut ...]
current 0 || isNOTenabled 0 == 0 (isCurrent() at /home/mrmara/workspace/src
/navigation2/nav2_costmap_2d/src/layered_costmap.cpp:275)
[controller_server-1] [INFO] [nav2_costmap_2d]: LayeredCostmap is current: 0 (isCu
rrent() at /home/mrmara/workspace/src/navigation2/nav2_costmap_2d/src/layer
ed_costmap.cpp:280)
[controller_server-1] [DEBUG] [icn04.controller_server]: Waiting for costmap to be
 valid (computeControl() at /home/mrmara/workspace/src/navigation2/nav2_con
troller/src/controller_server.cpp:408)
```
</details>

Does this makes sense to you?

SteveMacenski, 2024-02-22:
Please format your logs as you might see them in a terminal, I'm not sorting through that single line of thousands of unorganized messages.

>  The controller server remains stuck exactly where you said but I do not see any strange log on local costmap.

Great, it seems that your disabled obstacle layer isn't `current` when disabled. Sounds easy enough to fix, we need to have the state of `current_` be tempered by the state of `enabled_`, so that when disabled, that `current_` can still render to `true`. That fix should probably be applied to all the costmap layers as well

mrmara, 2024-02-26:
Please accept my apologies for bad formatting the logs.
What I wanted to say is that the logic you mentioned is already implemented inside the class ```LayeredCostmap```.
```
bool LayeredCostmap::isCurrent()
{
  current_ = true;
  for (vector<std::shared_ptr<Layer>>::iterator plugin = plugins_.begin();
    plugin != plugins_.end(); ++plugin)
  {
    current_ = current_ && ((*plugin)->isCurrent() || !(*plugin)->isEnabled());
    RCLCPP_DEBUG(
        rclcpp::get_logger("nav2_costmap_2d"),
        "Checking plugin %s, iscurrent %d || isNOTenabled %d == %d",
        (*plugin)->getName().c_str(), (*plugin)->isCurrent(), !(*plugin)->isEnabled(), current_);
  }
  for (vector<std::shared_ptr<Layer>>::iterator filter = filters_.begin();
    filter != filters_.end(); ++filter)
  {
    current_ = current_ && ((*filter)->isCurrent() || !(*filter)->isEnabled());
    RCLCPP_DEBUG(
        rclcpp::get_logger("nav2_costmap_2d"),
        "Checking filter %s, iscurrent %d || isNOTenabled %d == %d",
        (*filter)->getName().c_str(), (*filter)->isCurrent(), !(*filter)->isEnabled(), current_);
  }
  RCLCPP_DEBUG(
      rclcpp::get_logger("nav2_costmap_2d"),
      "LayeredCostmap is current: %d", current_);
  return current_;
}
```

I added some debug logs to investigate more the matter and apparently when the obstacle layer is disabled it makes all the other layers ```isCurrent()``` method returns False. I am going to debug a little more and get back.
Thank you very much for your help

SteveMacenski, 2024-02-26:
Huh that reads right to me - so please do! :smile:

SteveMacenski, 2024-03-22:
@mrmara any update?

mrmara, 2024-03-22:
I have done some tests immediately after opening the issue but I couldn't find anything useful. The truth is that I did not find enough time to spend on it. Please, let me update you by the end of the day.

mrmara, 2024-03-22:
I did some debug starting from the following observations:
- If obstacle_layer is disabled the costmap never become current [here](https://github.com/ros-planning/navigation2/blob/main/nav2_controller/src/controller_server.cpp#L468-L471 )
- Disabling instead inflation_layer, denoise_layer or keepout filter, the costmap correctly become current
- When the obstacle_layer is disabled each other layer never become current [here](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/layered_costmap.cpp#L260-L274)

So I saw were the variable ```current_``` is set to true in each layer and this happens to be in the method ```updateCosts```.

When I disable the obstacle_layer the```updateCosts``` method of each costmap plugin is never called because the [method LayeredCostmap::updateMap](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/layered_costmap.cpp#L133) returns I did some debug starting from the following observations:
- If obstacle_layer is disabled the costmap never become current [here](https://github.com/ros-planning/navigation2/blob/main/nav2_controller/src/controller_server.cpp#L468-L471 )
- Disabling instead inflation_layer, denoise_layer or keepout filter, the costmap correctly become current
- When the obstacle_layer is disabled each other layer never become current [here](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/layered_costmap.cpp#L260-L274)

So I saw were the variable ```current_``` is set to true in each layer and this happens to be in the method ```updateCosts```.

When I disable the obstacle_layer the```updateCosts``` method of each costmap plugin is never called because the [here](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/layered_costmap.cpp#L212).
This happens because the method ```updateBounds``` of every layer, except for Obstacle and Voxel, has not a control on the variable ```enabled_```. Obstacle and Voxel layer, instead, check ```enabled_``` and return immediately making [this check](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/src/layered_costmap.cpp#L212) return.

Does this make sense? Is this something wanted?

SteveMacenski, 2024-03-22:
What I'm hearing is that there is nothing to fix here and that you need something which sets the sizing -- so if you only have 1 obstacle layer and nothing else, it can't be current with only the inflation layer (i.e. no static layer)?

mrmara, 2024-03-22:
Understood! Thanks for explaining.

My local costmap is composed by keepout_filter, denoise_layer, inflation_layer and obstacle_layer. Occasionally I would like to disable obstacle layer and navigate with the remaining filters. This is not possible due to the reason explained above. For now I solved adding a "fake_obstacle_layer" which is basically doing nothing. I would really like to get rid of it, do you have any suggestions?

SteveMacenski, 2024-03-22:
Ah got it, that makes sense, you need something to actually set the size so that the other layers can operate based off of it with some kind of update. 

You are aware that by disabling the only obstacle layer you have in your system that you're navigating completely blind, yes? Honestly, even though that wasn't the intention of `enabled_`, that actually feels like its the correct output action to reject updates since you have not a map to navigate blind based off of nor sensor data to tell you that you're in clear space to continue moving within.

What's your expected behavior here or rationale for why this is a valid state to be in? I can definitely appreciate wanting to disable a sensor at some time, but if you don't then have other sensors or a map (or anything) then you're attempting to plan / control in a completely undefined space. I would expect that if you were disabling sensors, that then you'd have something like a map or other sensors available to make sure what you're doing is at least to some definition known to be valid

mrmara, 2024-03-22:
> What's your expected behavior here or rationale for why this is a valid state to be in?
Basically I want to navigate relaying only on nav2_collision_monitor hence I need to run only keepout_filter and inflation_layer to avoid planning through mapped obstacles, anyway I see your reasoning and it makes perfectly sense.

Thank again for your time.

SteveMacenski, 2024-03-22:
There are no mapped obstacles for the inflation layer to apply if you didn't include the static layer - which you haven't mentioned using in the local costmap

mrmara, 2024-03-22:
I use the inflation layer to inflate the keepout filter that includes also some mapped obstacles

SteveMacenski, 2024-03-22:
Well, I suppose in that way you're misusing the API, which I suppose is your right but is also what is causing the issue :wink: 

I guess I'm going to close this as `wontfix`. I can't find any reasonable solution in software, the solution is the use some (obstacle, static, etc) layer that sets the size and data so that the robot isn't driving blind. Currently you're driving blind without a map nor sensor data which is not a supported nor safe case.

closed 2024-03-22 by SteveMacenski