ros-navigation/navigation2 #4141: SmacHybrid planner can crash to segmentation fault

jonipol, 2024-02-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble (latest ros:humble container)
- Version or commit hash:
  - a3cdbbf7d0a9be9ec50f876ab365a45eff2214d9 + minimal changes to make the rolling version of nav2 to work on Humble (no changes to planner(s) code)
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
This part I am still uncertain. The issue keep happening seemingly randomly during navigation. Have not been able to pinpoint the exact reason/case as I do not yet fully understand what is going on in the SmacHybridPlanner. Sometimes the segmentation fault happens quite soon after starting navigating. Sometimes it takes many many navigations. (Had my simulation giving consecutive goals for 95 minutes before I got the latest seg fault for the log to include in this issue.)

#### Expected behavior
Planner keeps running :grin: 

#### Actual behavior
Planner server crashes to a segmentation fault.
<details>
<summary>Log</summary>

//     Logs I have added
[planner_server-2] D: 1.22173 is smaller than sqrt(2)
[planner_server-2] From x: 603.254 From y: 180.122 From yaw: 5.67232
[planner_server-2] To x: 604 To y: 180 To yaw: 0
//     My logs end
[planner_server-2] Stack trace (most recent call last) in thread 1430:
[planner_server-2] #31   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7ffaaf4aaee7, in 
[planner_server-2] #30   Source "/usr/include/c++/11/mutex", line 712, in _Prepare_execution<std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*&&, bool*&&)::<lambda()> > [0x7ffaafe2eb2b]
[planner_server-2]         709: 	// Store address in thread-local pointer:
[planner_server-2]         710: 	__once_callable = std::__addressof(__c);
[planner_server-2]         711: 	// Trampoline function to invoke the closure via thread-local pointer:
[planner_server-2]       > 712: 	__once_call = [] { (*static_cast<_Callable*>(__once_callable))(); };
[planner_server-2]         713:       }
[planner_server-2]         714: 
[planner_server-2]         715:     ~_Prepare_execution()
[planner_server-2] #29   Source "/usr/include/c++/11/mutex", line 712, in operator() [0x7ffaafe2eb16]
[planner_server-2]         709: 	// Store address in thread-local pointer:
[planner_server-2]         710: 	__once_callable = std::__addressof(__c);
[planner_server-2]         711: 	// Trampoline function to invoke the closure via thread-local pointer:
[planner_server-2]       > 712: 	__once_call = [] { (*static_cast<_Callable*>(__once_callable))(); };
[planner_server-2]         713:       }
[planner_server-2]         714: 
[planner_server-2]         715:     ~_Prepare_execution()
[planner_server-2] #28   Source "/usr/include/c++/11/mutex", line 776, in operator() [0x7ffaafe22ddd]
[planner_server-2]         773:     {
[planner_server-2]         774:       // Closure type that runs the function
[planner_server-2]         775:       auto __callable = [&] {
[planner_server-2]       > 776: 	  std::__invoke(std::forward<_Callable>(__f),
[planner_server-2]         777: 			std::forward<_Args>(__args)...);
[planner_server-2]         778:       };
[planner_server-2] #27   Source "/usr/include/c++/11/bits/invoke.h", line 96, in __invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*> [0x7ffaafe2eade]
[planner_server-2]          93:       using __result = __invoke_result<_Callable, _Args...>;
[planner_server-2]          94:       using __type = typename __result::type;
[planner_server-2]          95:       using __tag = typename __result::__invoke_type;
[planner_server-2]       >  96:       return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
[planner_server-2]          97: 					std::forward<_Args>(__args)...);
[planner_server-2]          98:     }
[planner_server-2] #26   Source "/usr/include/c++/11/bits/invoke.h", line 74, in __invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()>*, bool*> [0x7ffaafe3abcc]
[planner_server-2]          71:     __invoke_impl(__invoke_memfun_deref, _MemFun&& __f, _Tp&& __t,
[planner_server-2]          72: 		  _Args&&... __args)
[planner_server-2]          73:     {
[planner_server-2]       >  74:       return ((*std::forward<_Tp>(__t)).*__f)(std::forward<_Args>(__args)...);
[planner_server-2]          75:     }
[planner_server-2]          76: 
[planner_server-2]          77:   template<typename _Res, typename _MemPtr, typename _Tp>
[planner_server-2] #25   Source "/usr/include/c++/11/future", line 571, in _M_do_set [0x7ffaafe1bc15]
[... 197 lines cut ...]
[planner_server-2]         258:       // If any element is above the comfortable cost limit, check edge cases:
[planner_server-2]       > 259:       // (1) Check if goal is in greater than max_cost space requiring
[planner_server-2]         260:       //  entering it, but only entering it on final approach, not in-and-out
[planner_server-2]         261:       // (2) Checks if goal is in normal space, but enters costed space unnecessarily
[planner_server-2]         262:       //  mid-way through, skirting obstacle or in non-globally confined space
[planner_server-2] Segmentation fault (Address not mapped to object [(nil)])
[ERROR] [planner_server-2]: process has died [pid 205, exit code -11, cmd '/karelics_workspace/karelics_brain/nav2_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server -p use_sim_time:=True --params-file /tmp/tmp3_qleyov --params-file /tmp/tmp6ufp_bh7 -r /tf:=tf -r /tf_static:=tf_static'].

Note: On the last part of the trace the line with  `>` seems to point to wrong line. On earlier logs it was the line 257 the [if clause with dereferencing the max_element](https://github.com/ros-planning/navigation2/blob/55c952d953ce4dd9a141844d4c80894968a5bb56/nav2_smac_planner/src/analytic_expansion.cpp#L251). Not sure why. Could be backward_ros and changing the file on runtime combination or something. 
</details>

#### Additional information

I have been investigating this now for a bit. According to [StackOverflow comment](https://stackoverflow.com/a/41289894/8025833) dereferencing the iterator returned by `std::max_element` when the input array is empty, can lead into segmentation fault. Running `std::max_element(EmptyArray)` returns `EmptyArray.end()`. I believe that is what happens in this case.

To have the `node_costs` array to be empty the `num_iterations` has to be 0. There is a comment on line [196](https://github.com/ros-planning/navigation2/blob/55c952d953ce4dd9a141844d4c80894968a5bb56/nav2_smac_planner/src/analytic_expansion.cpp#L196) about the possibility but I fail to notice any action taken for that. Basically if distance between from and to poses is less than sqrt(2) we get the zero. 

For myself I fixed this by returning early in similar fashion as on line [186](https://github.com/ros-planning/navigation2/blob/55c952d953ce4dd9a141844d4c80894968a5bb56/nav2_smac_planner/src/analytic_expansion.cpp#L186). Simple `if (d < sqrt_2) return AnalyticExpansionNodes();` and that seems to be working fine for me.

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2024-02-23:
> I have been investigating this now for a bit. According to [StackOverflow comment](https://stackoverflow.com/a/41289894/8025833) dereferencing the iterator returned by std::max_element when the [...]

The C++ docs that I usually reference have some remark if that would be a problem that I see aren't listed there. But just being a sane person knowing STL, yeah your expectation there would align with mine. I think the documentation just had that case missing. 

Can you adjust that to: 

```
auto max_cost_it = std::max_element(node_costs.begin(), node_costs.end());
if (max_cost_it != node_costs.end() && *node_costs > max_cost) {
```

Does it go away? If so, submit a PR to `main` and I'll merge + backport it to all the distributions. Thanks for the note, I agree with your analysis that this is the culprit. 

> For myself I fixed this by returning early in similar fashion as on line [...]

You can add that too! Why not both, I practice defensive programming.

jonipol, 2024-02-23:
Thanks for quick response and the feedback! I will do this once I am back from my trip on Monday.

SteveMacenski, 2024-02-26:
Merging imminent

closed 2024-02-26 by SteveMacenski