ros-navigation/navigation2 #4150: SmacPlannerHybrid segfault

BriceRenaudeau, 2024-02-27:
## Bug report

Sometimes, the planner_server dies because of a segfault.
This seems to happen only on very small paths using `SmacPlannerHybrid`.
It may be related to this commit: 
https://github.com/ros-planning/navigation2/commit/e2781e57fd445712d7687d28449a7495a33e495f

```
thread 24 "planner_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffa22e0640 (LWP 265292)]
0x00007fffb04dd878 in nav2_smac_planner::AnalyticExpansion<nav2_smac_planner::NodeHybrid>::getAnalyticPath(nav2_smac_planner::NodeHybrid* const&, nav2_smac_planner::NodeHybrid* const&, std::function<bool (unsigned int const&, nav2_smac_planner::NodeHybrid*&)> const&, std::shared_ptr<ompl::base::StateSpace> const&) () from /opt/ros/iron/lib/libnav2_smac_planner.so
(gdb) bt
#0  0x00007fffb04dd878 in nav2_smac_planner::AnalyticExpansion<nav2_smac_planner::NodeHybrid>::getAnalyticPath(nav2_smac_planner::NodeHybrid* const&, nav2_smac_planner::NodeHybrid* const&, std::function<bool (unsigned int const&, nav2_smac_planner::NodeHybrid*&)> const&, std::shared_ptr<ompl::base::StateSpace> const&) () from /opt/ros/iron/lib/libnav2_smac_planner.so
#1  0x00007fffb04de0a8 in nav2_smac_planner::AnalyticExpansion<nav2_smac_planner::NodeHybrid>::tryAnalyticExpansion(nav2_smac_planner::NodeHybrid* const&, nav2_smac_planner::NodeHybrid* const&, std::function<bool (unsigned int const&, nav2_smac_planner::NodeHybrid*&)> const&, int&, int&) () from /opt/ros/iron/lib/libnav2_smac_planner.so
#2  0x00007fffb04ca27b in nav2_smac_planner::AStarAlgorithm<nav2_smac_planner::NodeHybrid>::createPath(std::vector<nav2_smac_planner::NodeHybrid::Coordinates, std::allocator<nav2_smac_planner::NodeHybrid::Coordinates> >&, int&, float const&, std::vector<std::tuple<float, float, float>, std::allocator<std::tuple<float, float, float> > >*) () from /opt/ros/iron/lib/libnav2_smac_planner.so
#3  0x00007fffb04abc28 in nav2_smac_planner::SmacPlannerHybrid::createPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&) () from /opt/ros/iron/lib/libnav2_smac_planner.so
#4  0x00007ffff7f1dad0 in nav2_planner::PlannerServer::getPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/iron/lib/libplanner_server_core.so
#5  0x00007ffff7f24e9a in nav2_planner::PlannerServer::computePlan() () from /opt/ros/iron/lib/libplanner_server_core.so
#6  0x00007ffff7f4ad53 in nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::work() () from /opt/ros/iron/lib/libplanner_server_core.so
#7  0x00007ffff7f4b941 in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void> >::_M_invoke(std::_Any_data const&) () from /opt/ros/iron/lib/libplanner_server_core.so
#8  0x00007ffff7f2811d in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) () from /opt/ros/iron/lib/libplanner_server_core.so
#9  0x00007ffff716cee8 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#10 0x00007ffff7f2c3d1 in std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void>::_M_run() () from /opt/ros/iron/lib/libplanner_server_core.so
#11 0x00007ffff73d7253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#12 0x00007ffff7167ac3 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
#13 0x00007ffff71f9a40 in ?? () from /lib/x86_64-linux-gnu/libc.so.6
```

**Required Info:**

- Operating System:
  - Ubuntu 22.04 -->
- ROS2 Version:
  - Iron -->
- Version or commit hash:
  - source "main" -->
- DDS implementation:
  - cyclone -->

#### Steps to reproduce the issue
Request a small path planning using smac_planner_hybrid

#### Additional information

When compiling in debug mode, the bug doesn't happen and when I add debug `cout` in the function, it doesn't happen either.
 It looks like a race condition or concurrent access.

SteveMacenski, 2024-02-27:
Could it be related to https://github.com/ros-planning/navigation2/pull/4147's fix? That looks about the right spot and same kind of issue that was just patched. Take a look and let me know if that fixes your issue!

BriceRenaudeau, 2024-02-28:
Thanks for your quick reply.
Yes, I think it will solve the issue. I posted this issue just after you closed the other one #4141

closed 2024-02-28 by BriceRenaudeau