ros-navigation/navigation2 #4153: Robot drives near keepout zones, and stops when "incidentally" enters the keepout zone

IqbalBintangsyah, 2024-02-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - apt binary

#### Steps to reproduce issue
I'm using DWBLocalPlanner on a rectangular robot.
The robot needs to go from point A to point B, and in between, the robot has to turn right to avoid keepout zones (Picture below)
[image]

This is quite identical with this issue https://github.com/ros-planning/navigation2/issues/3102 , I've tried adding inflatioin layer, but the problem persists.

Parameter file:
```yaml
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_link"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 4.0
    laser_max_range: 12.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 100
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.1
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.9
    z_max: 0.1
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan
    set_initial_pose: True
    initial_pose:
      x: 20.144
      y: -1.114
      z: 0.0
      yaw: 1.615

amcl_map_client:
  ros__parameters:
    use_sim_time: True
[... 357 lines cut ...]
    use_sim_time: True
    smoothing_frequency: 20.0
    scale_velocities: False
    feedback: "OPEN_LOOP"
    max_velocity: [0.26, 0.0, 1.0]
    min_velocity: [-0.26, 0.0, -1.0]
    max_accel: [2.5, 0.0, 3.2]
    max_decel: [-2.5, 0.0, -3.2]
    odom_topic: "odom"
    odom_duration: 0.1
    deadband_velocity: [0.0, 0.0, 0.0]
    velocity_timeout: 1.0
```

#### Expected behavior
1. The robot drive further from the keepout zones. (Though I suppose the reason of why the robot drives near the keepout zone is because it's nearer.)
2. Even though the robot "enters" the keepout zones, I want it to still drive forward instead of stopping.

#### Actual behavior
The robot generate path that is very close with the keepout zone, and eventually enters it, and get stuck there.

#### Additional information
I have tried adding inflation layor as suggested here https://github.com/ros-planning/navigation2/issues/3102#issuecomment-1225494691 , the robot drives furthere from the keepout zone, but somehow still managed to get stuck in the keepout zone. If I make the inflation layer bigger, the robot won't e able to move along the narrow path.

Thank you

SteveMacenski, 2024-02-28:
Please suggest a change / design that would meet your needs! I'm open ears.

IqbalBintangsyah, 2024-02-28:
Hello!

I'm currently having trouble with the path planning and inflation layer. The path generated was like this

[image]

As you can see, the path (green line) is generated to pass the inflation layer (blue), which, I thought it is normal if there is no enough space for the robot. But then if the robot enter the inflation layer, it stops.

[image]

and the terminal says that the robot is on lethal area.

Is inflation layer supposed to be a lethal area? I tried to change `cost_scaling_factor` parameter to make the inflation layer "less lethal" but it has no effect (the robot still stops upon contact with inflation layer).

I'd suggest the robot to create path a little bit further from inflation area, and for the robot to not stop if it enters inflation area.

Thank you

SteveMacenski, 2024-02-28:
Please ask on Robotics Stack Exchange, this is not the right place to ask for configuration / concept help. If you'd like to discuss the keepout method, we can continue with that in this ticket, but we don't blend topics.

IqbalBintangsyah, 2024-02-28:
Oh okay! Thank you! I posted the question [here](https://robotics.stackexchange.com/questions/109618/robot-drives-near-keepout-zones-and-stops-when-incidentally-enters-the-keepou)

I'm going to close this ticket here

closed 2024-02-28 by IqbalBintangsyah