ros-navigation/navigation2 #4154: resetObstacleHeuristic segfaults on goal not in costmap

Aposhian, 2024-02-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
  - ros-humble-nav2-planner	1.1.12-1jammy.20240126.072321
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Request a goal that is outside the costmap.

#### Expected behavior
Probably should reject the goal, and log an error.

#### Actual behavior
Segfault with the following bactkrace
```
#0  0x00007f8da01f69e4 in nav2_smac_planner::NodeHybrid::resetObstacleHeuristic
    (costmap=<optimized out>, start_x=@0x7f8d0c8a0864: 210373280, 
    start_y=@0x7f8d0c8a0860: 32619, goal_x=@0x7f8d0c8a090c: 210373280, 
    goal_y=@0x7f8d0c8a0908: 32619)
    at /usr/include/c++/11/bits/stl_vector.h:1043
#1  0x00007f8da01e9889 in nav2_smac_planner::AStarAlgorithm<nav2_smac_planner::NodeHybrid>::setGoal (this=0x7f8d2c005e60, mx=@0x7f8d0c8a090c: 210373280, 
    my=@0x7f8d0c8a0908: 32619, dim_3=@0x7f8d0c8a0904: 45)
    at ./src/a_star.cpp:186
#2  0x00007f8da01d5bc6 in nav2_smac_planner::SmacPlannerHybrid::createPlan (
    this=0x7f8d2c0657f0, start=..., goal=...)
    at /usr/include/c++/11/bits/unique_ptr.h:173
#3  0x00007f8dc8100b23 in nav2_planner::PlannerServer::getPlan (
    this=0x55820b0e66a0, start=..., goal=..., planner_id=...)
    at ./src/planner_server.cpp:540
#4  0x00007f8dc8105ff3 in nav2_planner::PlannerServer::computePlan (
    this=0x55820b0e66a0) at ./src/planner_server.cpp:492
```

#### Additional information
I think the problem is that `goal_index` is computed, but never bounds checked against the costmap.
```
obstacle_heuristic_lookup_table[goal_index] = -0.00001f;
```
Although I haven't actually confirmed that in debug mode.

SteveMacenski, 2024-02-28:
> If it is due to requesting out of bounds, then I'm not sure why this code doesn't already catch it:

```
  if (!costmap->worldToMap(goal.pose.position.x, goal.pose.position.y, mx, my)) {
    throw nav2_core::GoalOutsideMapBounds(
            "Goal Coordinates of(" + std::to_string(goal.pose.position.x) + ", " +
            std::to_string(goal.pose.position.y) + ") was outside bounds");
  }
```

I 100% agree, can you see why that is? I don't check bounds after the initial transition to map x,y coordinates because it shouldn't be necessary. While a traceback to find the exact line in `resetObstacleHeuristic` that fails would be great, I think the better use of time is seeing how we even got past this check. When you say "well out of bounds", do you mean multiples on the robot's diameter, or just a few meters?

Aposhian, 2024-02-28:
By well out of bounds, I mean that the costmap should only have 1000x1000 cells, but from the stacktrace it is requesting x=210373280, and y=32619.

SteveMacenski, 2024-02-28:
Yikes. Yeah. Please do look into that and let me know. I could dig into it too, but you have the reproduction case handy and I think that's going to be > 50% of the total debugging time just being able to trigger it. If you have questions or get into trouble, let me know and if you can give me a way to trigger it on the TB3 demo or something else I can setup, let me know

Aposhian, 2024-02-28:
Sure, I anticipate that I'll track down more info as I go. Just wanted to get the issue going to see if anyone else sees this behavior.

SteveMacenski, 2024-03-28:
@Aposhian I spent a little bit of time today looking into this. Humble does not appear to have the:

```
  if (!costmap->worldToMap(goal.pose.position.x, goal.pose.position.y, mx, my)) {
    throw nav2_core::GoalOutsideMapBounds(
            "Goal Coordinates of(" + std::to_string(goal.pose.position.x) + ", " +
            std::to_string(goal.pose.position.y) + ") was outside bounds");
  }

``

logic. Adding it here: https://github.com/ros-planning/navigation2/pull/4224 that should resolve your problem

closed 2024-03-28 by SteveMacenski